Ultimaker/Cura #3107: Slowly start of program

scharrin2, 2018-01-08:
Hello,

the startup of Cura takes very long.

When showing the screen "Loading devices" it takes about 10 seconds until the next screen is shown.
I am using a TEVO Tarantula, which is not connected to the computer. I am only creating the gcode files and save them on SD card. Perhaps Cura is searching for the direct connection to the printer.
Is there an option to not searching for the connection? 

**Application Version**
3.1.0 (newest version - downloaded today)

**Platform**
OS X El Capitan 10.11.6

ChrisTerBeke, 2018-01-08:
This is a know issue. This will be fixed in Cura 3.2 which is due in a few weeks.

closed 2018-01-08 by ChrisTerBeke

fieldOfView, 2018-01-08:
For the record, this has nothing to do with Cura trying to connect to your printer, but more with Cura loading printer definitions, quality profiles and materials for all combinations of printers, qualities and materials known to man (or at least known to Cura).

scharrin2, 2018-01-08:
But is it necessary to load all printer definitions when I start Cura? I think it would be better to load this data in the settings dialog when adding a new printer.

fieldOfView, 2018-01-08:
No, that is what is fixed in Cura 3.2