Ultimaker/Cura #3108: Open SVG File

DenVog, 2018-01-08:
Cura should have the ability to open files saved in the .svg file format.

**Application Version**
3.1.0

**Platform**
macOS High Sierra 10.13.2

**Steps to Reproduce**
1. File > Open File
2. Try to select a .svg file

**Actual Results**
Cura is unable to open a .svg file. So if you're working with a desktop imaging application such as Adobe Illustrator or Photoshop, you have to save to a bitmap format like .gif, .jpg, or .png in order for Cura to open it. These bitmap files create anti-aliasing of angles and rounded edges. These will in turn result in uneven side walls when Cura slices.

Alternatively, you can use your imaging application to save as .svg. This then requires an intermediate application such as Tinkercad to open the .svg and save it as .stl. Then go back to Cura to open the .stl and slice then save a gcode for Ultimaker. It's an additional set of steps that seem unnecessary. 

**Expected results**
Cura should recognize a .svg file as a valid file type to open, then launch the Convert Image... dialog.

**Additional Information**
If you're creating your original graphics or text in a desktop imaging application like Adobe Illustrator or Photoshop, you don't want to save those files as bitmap for slicing  if you care about 3D print quality.

ChrisTerBeke, 2018-01-17:
Sounds logical to add more image types that are popular.

One concern I have about SVG is that it not always contains a size (as it's vector), so we need a default solution for when that's the case (e.g. assume it's in mm).

fieldOfView, 2019-07-20:
This is implemented in a plugin, so the issue can be closed.

Ghostkeeper, 2019-07-22:
The [SVGToolpathReader](https://github.com/Ghostkeeper/SVGToolpathReader) loads the SVG files as g-code, if that's what you mean. It doesn't allow you to actually slice it as a volume, though that is something I may be implementing in the future.

closed 2023-07-13 by github-actions[bot]