Ultimaker/Cura #3110: Infill mesh influences skirt shape

smartavionics, 2018-01-09:
**Application Version**

master branch

**Platform**

Linux

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

Add model with rectangular base to build plate. Add infill mesh that extends past outline of model. Define skirt with skirt distance of 1mm.

**Actual Results**

Skirt is influenced by the infill mesh so it doesn't follow the rectangular outline of the mode.

**Expected results**

Skirt should follow rectangular outline of first layer of model.

**Additional Information**

[image]

Ghostkeeper, 2018-01-25:
I can't reproduce this. Here is a cube with a skirt, and an infill mesh punching through it.

[image]

The skirt does not go around the infill mesh, nor does it seem to be influenced by it.

Was it only happening on that specific model?

smartavionics, 2018-01-26:
Sorry, I blundered, it's not related to the infill mesh, it's something related to where the support towers would be placed. Here's the same model with the infill mesh removed and first support without towers and then support with towers. You can see the skirt bends around where the tower bases would be. Let's forget this. Sorry to have wasted your time.

[image]

[image]

closed 2018-01-26 by smartavionics

Ghostkeeper, 2018-02-13:
That is also buggy, but that would make this a duplicate of https://github.com/Ultimaker/CuraEngine/issues/544 .