Ultimaker/Cura #3111: infill mesh influences layer count in GUI and engine

smartavionics, 2018-01-09:
**Application Version**

master branch

**Platform**

Linux

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

Add model and infill mesh that extends above the height of the model.

**Actual Results**

The layer slider max value is greater than the number of layers.

**Expected results**

The layer slider should stop when it reaches the top layer of the model (ignoring the infill mesh).

**Additional Information**

The backend is also being influenced by the infill mesh because it reports:
````
2018-01-09 11:47:56,748 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] [WARNING] There are empty layers (or layers with empty extruder plans) in the print! Temperature control and cross layer travel moves might suffer.
2018-01-09 11:47:56,748 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] [WARNING] There are empty layers (or layers with empty extruder plans) in the print! Temperature control and cross layer travel moves might suffer.

````

diegopradogesto, 2018-01-09:
I have seen that when I was working in the simulation view. Sometimes Cura receives empty layers at the end. That's why the horizontal (paths) slider has no handle, because there are no lines.

Ghostkeeper, 2018-01-25:
Yes, I could reproduce this bug too using two cubes.

closed 2018-04-15 by smartavionics