Ultimaker/Cura #3112: Cura has stopped functioning

Raveq, 2018-01-09:
**Application Version**

All except 2.3.1
**Platform**

Windows 10
**Qt**

**PyQt**

**Display Driver**

NVIDIA Quadro K2200

**Steps to Reproduce**
I try to open or save a file (always with menu option, if I dragg&dropp a file I can open it, but never save it)

**Actual Results**
 Cura crashes

**Expected results**

Save the file or let open a file
**Additional Information**

I can only work with Cura 2.3.1, is the only version that works for me.
[cura log.zip](https://github.com/Ultimaker/Cura/files/1615803/cura.log.zip)

nallath, 2018-01-17:
Unfortunately, we won't make any bug fixes or improvements to 2.3.1 any more.

closed 2018-01-17 by nallath

Ghostkeeper, 2018-01-25:
It seems that one of your machines has characters that 2.3.1 couldn't handle well. You might have to go into `C:/Users/<You>/AppData/Local/cura` and delete traces of a printer called "ultimaker2_extended_plus_fdmextruder_hq+piezas+peque%C3%B1as".