Ultimaker/Cura #3113: Cura 3.1 not using default print temp

luckyjay, 2018-01-09:
**Application Version**
3.1 from Ultimaker's website

**Platform**
Windows 10

**Steps to Reproduce**
Set a initial printing temperature that is not the default (210). Example, set 265 for ABS.

**Actual Results**
Cura does not honor the specified initial printing temperature. Printing starts at 210 in all cases.

**Expected results**
Printing should not start until the initial printing temperature setting is reached.

** Additional Notes **
The starting temp is being set via Start Gcode for my Lulzbot Mini:

M109 R{material_print_temperature}

I'm not sure where it's getting the {material_print_temperature} from, since the default print temp for ABS is 230, and in Cura I have all print temps set to 265. but that Gcode is being rendered as M109 R210.

Hopefully I'm not being a total noob here, but I don't remember this happening before with Cura.

fieldOfView, 2018-01-09:
Duplicate of #3106, #3061, #3032, #1296 and others. Will be fixed in Cura 3.2

luckyjay, 2018-01-09:
Sorry. My search skills must suck!

closed 2018-01-09 by luckyjay