Ultimaker/Cura #3115: Z Hop over top layers

Smithy101, 2018-01-10:
This may be more of an enhancement request than a bug.
I found that Cura doesn't lift the head while moving over top layers. (Z hop when retracted is ON, and retract at layer change is also ON) Instead of lifting the nozzle before starting the next layer OVER the next layer height, it goes to that height and then to the next starting point. Example, if current layer is 2.7mm and next layer is 3.00, it will raise nozzle to only 3.00 and not hop higher. (0.3mm res)

The result is a line where the nozzle goes across the top layer and puts a line through it.
This is even more noticeable when printing a flat surface with text on it. You can see where the nozzle comes from the corner over the top layer to start the text on the next layer.

Here is an example of the Gcode Cura generates, and how I modified the Gcode to eliminate the line.
Can this be a feature to make the nozzle stay away from the top layer? 
I move the retraction point and also forced it to hop over the top layer instead of moving the bit across it.
I'm getting MUCH better results with this gcode modification. (Yes, relative extrusion is also on)

**Example code before modification**
G0 F4800 X34.448 Y113.079
G1 F1800 X33.958 Y112.588 E0.03461
G0 F4800 X80.178 Y107.387  <--- This command puts the line on the top layer.
;TIME_ELAPSED:2447.165067
;LAYER:11
G1 F1500 E-4
G0 F4800 X80.178 Y107.387 Z3.6
;TYPE:WALL-INNER
G1 F1500 E4
G1 F3000 X73.695 Y107.393 E0.32344
G1 X73.696 Y111.556 E0.20769

**Fixed Gcode**
G0 F4800 X34.448 Y113.079
G1 F1800 X33.958 Y112.588 E0.03461    ;Last line of Layer 10
G1 F1500 E-4       ;Makes more sense to retract here
G1 F1800 X33.958 Y112.588 Z5     ;Just need Z a little over surface.
G0 F4800 X80.178 Y107.387          ;Now move over to new location
;TIME_ELAPSED:2447.165067
;LAYER:11
G0 F4800 X80.178 Y107.387 Z3.6
;TYPE:WALL-INNER
G1 F1500 E4
G1 F3000 X73.695 Y107.393 E0.32344
G1 X73.696 Y111.556 E0.20769

**Application Version**
3.1.0

**Platform**
Windows 10

**Steps to Reproduce**
Draw a box. 
Put text on top of box.
Cura will leave a line from the hot nozzle from corner of box over to text.

smartavionics, 2018-01-10:
I am guessing that you have combing set to "all". If so, have you tried setting combing to "no skin"?

Smithy101, 2018-01-12:
That didn't stop it from going right over the top skin.
Instead of coming in from the corner, it comes in from right above the text, but still doesn't lift over the top skin.
Here is the code with "No Skin" turned on.

G1 F1800 X33.958 Y121.073 E0.01806 <-- Last printed line in Layer 10
G1 F1500 E-4
G0 F4800 X31.723 Y121.073
G0 X31.735 Y123.553
G0 X58.942 Y123.565  <-- Would like to see the Z Hop right here...
G0 X58.942 Y113.203  <-- Goes straight to letter, but no lift of the nozzle.
;TIME_ELAPSED:3901.814676
;LAYER:11
G0 X58.942 Y113.203 Z3.6
;TYPE:WALL-INNER
G1 F1500 E4
G1 F3600 X63.14 Y113.203 E0.20944
G1 X63.14 Y113.505 E0.01507
G1 X58.942 Y113.505 E0.15813
G1 X58.456 Y113.505 E0.02425

smartavionics, 2018-01-12:
Hi, it would help if you code post a link to a sample gcode file that has this problem. It makes it much easier to understand if I can step through the gcode and see all the settings, etc.

smartavionics, 2018-01-13:
Hi, I just sliced your STL file using one of my stock profiles and looking at the gcode, I can see that it z-hops between each of the letters (including the initial travel from the corner.)
[image]
I shall see if I can determine the difference between your profile and mine. Will be back...

smartavionics, 2018-01-13:
FYI, these are the settings I used:

````
;SETTING_3 {"extruder_quality": ["[general]\\nversion = 2\\nname = Kossel XL (ho
;SETTING_3 t bed)\\ndefinition = kossel_xl\\n\\n[metadata]\\nquality_type = norm
;SETTING_3 al\\nextruder = fdmextruder\\ntype = quality_changes\\nsetting_versio
;SETTING_3 n = 4\\n\\n[values]\\nacceleration_wall = None\\ncool_fan_full_at_hei
;SETTING_3 ght = 3\\ncool_fan_min_at_height = 1\\ncool_fan_speed = 80.0\\ncool_f
;SETTING_3 an_speed_0 = 20\\ncool_fan_speed_min = 40.0\\ncool_min_layer_time = 5
;SETTING_3 \\ncool_min_layer_time_fan_speed_max = 15\\ncool_min_speed = 15\\ndef
;SETTING_3 ault_material_print_temperature = 185\\nexpand_skins_expand_distance 
;SETTING_3 = 2\\nfill_perimeter_gaps = nowhere\\ninfill_before_walls = False\\ni
;SETTING_3 nfill_line_width = 0.8\\ninfill_overlap = 10\\ninfill_pattern = zigza
;SETTING_3 g\\ninfill_sparse_density = 25\\njerk_wall = 5\\nline_width = 0.5\\nm
;SETTING_3 aterial_diameter = 1.75\\nmaterial_flow_layer_0 = 80\\nmaterial_print
;SETTING_3 _temperature = =default_material_print_temperature\\nmaterial_print_t
;SETTING_3 emperature_layer_0 = 190\\nmax_skin_angle_for_expansion = 40\\noptimi
;SETTING_3 ze_wall_printing_order = True\\nretraction_amount = 2\\nretraction_ex
;SETTING_3 tra_prime_amount = 0.1\\nretraction_hop = 0.5\\nretraction_hop_enable
;SETTING_3 d = True\\nretraction_min_travel = 2.5\\nretraction_prime_speed = =re
;SETTING_3 traction_speed\\nretraction_retract_speed = =retraction_speed\\nretra
;SETTING_3 ction_speed = 20\\nskin_outline_count = 0\\nskin_overlap = None\\nski
;SETTING_3 rt_brim_minimal_length = 500\\nskirt_gap = 8\\nspeed_infill = 60\\nsp
;SETTING_3 eed_layer_0 = 25.0\\nspeed_print = 60\\nspeed_print_layer_0 = 40.0\\n
;SETTING_3 speed_topbottom = 40.0\\nspeed_travel = 300\\nspeed_travel_layer_0 = 
;SETTING_3 300.0\\nspeed_wall_0 = 50.0\\nspeed_wall_x = 60.0\\nspeed_walls_layer
;SETTING_3 _0 = 20.0\\nsupport_angle = 60\\nsupport_infill_rate = 25\\nsupport_i
;SETTING_3 nterface_enable = True\\nsupport_pattern = zigzag\\nsupport_xy_distan
;SETTING_3 ce = 1\\nsupport_z_distance = 0.3\\ntop_bottom_pattern = lines\\ntop_
;SETTING_3 bottom_thickness = 0.6\\nwall_0_wipe_dist = 0\\nwall_thickness = 1\\n
;SETTING_3 z_seam_relative = True\\nz_seam_x = 100\\nz_seam_y = 100\\n\\n"], "gl
;SETTING_3 obal_quality": "[general]\\nversion = 2\\nname = Kossel XL (hot bed)\
;SETTING_3 \ndefinition = kossel_xl\\n\\n[metadata]\\nquality_type = normal\\nty
;SETTING_3 pe = quality_changes\\nsetting_version = 4\\n\\n[values]\\naccelerati
;SETTING_3 on_enabled = False\\nadaptive_layer_height_variation = 0.15\\nadhesio
;SETTING_3 n_type = skirt\\njerk_enabled = False\\nlayer_height = 0.2\\nlayer_he
;SETTING_3 ight_0 = 0.25\\nmaterial_bed_temperature = 45\\nmaterial_bed_temperat
;SETTING_3 ure_layer_0 = =resolveOrValue('material_bed_temperature')\\nretractio
;SETTING_3 n_combing = noskin\\nspeed_slowdown_layers = 1\\nsupport_enable = Fal
;SETTING_3 se\\n\\n"}
````

smartavionics, 2018-01-13:
Ah, the 3mf didn't contain your profile so I don't know what is different. You could post the settings from the bottom of your gcode like I did above.

Smithy101, 2018-01-13:
Strange.
It's not hopping for me before making a layer change. 
Would like to see it raise above the next layer when moving over the top skin. Example, if skin is at 3.00mm, then raise nozzle to X distance for hop over skin before starting layer over that skin. In my case, I made it 1.4mm higher to make sure the hit nozzle had no effect on skin.

Settings..
;SETTING_3 {"global_quality": "[general]\\nversion = 2\\nname = empty\\ndefiniti
;SETTING_3 on = custom\\n\\n[metadata]\\nquality_type = draft\\ntype = quality_c
;SETTING_3 hanges\\n\\n[values]\\nacceleration_enabled = False\\nadhesion_type =
;SETTING_3  skirt\\ndraft_shield_dist = 5\\ndraft_shield_height_limitation = lim
;SETTING_3 ited\\njerk_enabled = False\\nlayer_height = 0.3\\nmaterial_bed_tempe
;SETTING_3 rature = 70\\nrelative_extrusion = True\\nretraction_combing = all\\n
;SETTING_3 travel_retract_before_outer_wall = True\\n\\n", "extruder_quality": [
;SETTING_3 "[general]\\nversion = 2\\nname = empty\\ndefinition = custom\\n\\n[m
;SETTING_3 etadata]\\nquality_type = draft\\nextruder = custom_extruder_1\\ntype
;SETTING_3  = quality_changes\\n\\n[values]\\nacceleration_print = 1500\\nbottom
;SETTING_3 _skin_expand_distance = 0.5\\ncool_fan_full_layer = 5\\ncool_fan_spee
;SETTING_3 d = 25\\ndefault_material_print_temperature = 215\\nironing_only_high
;SETTING_3 est_layer = True\\nironing_pattern = concentric\\njerk_print = 15\\nm
;SETTING_3 aterial_final_print_temperature = 215\\nmaterial_initial_print_temper
;SETTING_3 ature = 215\\nmeshfix_extensive_stitching = True\\nretract_at_layer_c
;SETTING_3 hange = True\\nretraction_amount = 4\\nretraction_hop_enabled = True\
;SETTING_3 \nretraction_hop_only_when_collides = True\\nskirt_brim_speed = 25\\n
;SETTING_3 skirt_line_count = 4\\nspeed_layer_0 = 25\\nspeed_print = 40\\nspeed_
;SETTING_3 topbottom = 30\\nspeed_travel = 80\\ntop_bottom_thickness = 1.2\\nwal
;SETTING_3 l_thickness = 1.2\\n\\n", "[general]\\nversion = 2\\nname = empty\\nd
;SETTING_3 efinition = custom\\n\\n[metadata]\\nquality_type = normal\\nextruder
;SETTING_3  = custom_extruder_2\\ntype = quality_changes\\n\\n[values]\\n\\n", "
;SETTING_3 [general]\\nversion = 2\\nname = empty\\ndefinition = custom\\n\\n[me
;SETTING_3 tadata]\\nquality_type = normal\\nextruder = custom_extruder_3\\ntype
;SETTING_3  = quality_changes\\n\\n[values]\\n\\n", "[general]\\nversion = 2\\nn
;SETTING_3 ame = empty\\ndefinition = custom\\n\\n[metadata]\\nquality_type = no
;SETTING_3 rmal\\nextruder = custom_extruder_4\\ntype = quality_changes\\n\\n[va
;SETTING_3 lues]\\n\\n", "[general]\\nversion = 2\\nname = empty\\ndefinition = 
;SETTING_3 custom\\n\\n[metadata]\\nquality_type = normal\\nextruder = custom_ex
;SETTING_3 truder_5\\ntype = quality_changes\\n\\n[values]\\n\\n", "[general]\\n
;SETTING_3 version = 2\\nname = empty\\ndefinition = custom\\n\\n[metadata]\\nqu
;SETTING_3 ality_type = normal\\nextruder = custom_extruder_6\\ntype = quality_c
;SETTING_3 hanges\\n\\n[values]\\n\\n", "[general]\\nversion = 2\\nname = empty\
;SETTING_3 \ndefinition = custom\\n\\n[metadata]\\nquality_type = normal\\nextru
;SETTING_3 der = custom_extruder_7\\ntype = quality_changes\\n\\n[values]\\n\\n"
;SETTING_3 , "[general]\\nversion = 2\\nname = empty\\ndefinition = custom\\n\\n
;SETTING_3 [metadata]\\nquality_type = normal\\nextruder = custom_extruder_8\\nt
;SETTING_3 ype = quality_changes\\n\\n[values]\\n\\n"]}

Smithy101, 2018-01-13:
[Text Test (2).zip](https://github.com/Ultimaker/Cura/files/1628634/Text.Test.2.zip)
Full Gcode..

smartavionics, 2018-01-13:
> ;SETTING_3 rature = 70\nrelative_extrusion = True\nretraction_combing = all\n

You have combing set to all in your profile, it needs to be "no skin" (or off completely).

Smithy101, 2018-01-13:
Your correct. That did fix this test piece when completely turned off.
Sorry, I turn that on/off depending on what I'm slicing.

This would still not fix a test piece if I needed coming turning on for parts of the piece, but needed it to hop over the face for other parts. Example, a more complex print that also had text at the top.

Would be nice if there was a specific setting to z-hop over top skin even when coming is turned on. Or an extra setting in the coming to z-hop over top skin.

smartavionics, 2018-01-13:
Yes, I think the combing causes a lot of troubles. People are complaining about it a lot. Personally, I only ever use the "no skin" setting or none at all for runny materials like PETG.

Smithy101, 2018-01-13:
Now if only this fillenium malcon would finish printing.... lol 
Gotta have fun with the printer as well.

Ghostkeeper, 2018-01-28:
Would it make sense to change the current option "No Skin" to "No Top Surface"?

At least I'd like to rename it to "Not in Skin" or something of the likeness.

smartavionics, 2018-01-28:
Personally, I would never use combing over any skin because even if it wasn't the top surface, the dribble is still going to be visible as a bump in the skin layers above.

filipgoc, 2018-03-14:
@Smithy101 has a point. It could would be useful to have a setting that does what people most likely want, and not a somewhat cryptic combination of other settings that achieve this. 

From the user's perspective, most people want to
> Z Hop over the top layer

They don't think about it in negative terms as in
> Comb all layers (meaning do no Z hop - even that is not self evident) but do NOT comb the skin (=top layer, also not self evident to newbies)

closed 2018-07-31 by ianpaschal

nh2, 2019-03-13:
Here's a video I found that shows what settings to activate to ensure z-hop everywhere:

https://www.youtube.com/watch?v=FZQqGN0_tLY

I believe that the difference it makes in the UI is thin-dark-blue travel lines vs thicker-light-blue travel lines that do z-hop (settings highlighted red; also note the 15 minutes increased printing time):

##### No Z-hop on top surface layer

[image]

##### Z-hop on top surface layer

[image]

Ghostkeeper, 2019-03-14:
In the interface, the darker blue lines indicate travel moves without retraction. The lighter blue lines indicate retracted travel moves. If you have the Z Hop When Retracted setting enabled and the Z Hop Only Over Printed Parts disabled, every light-blue line should be a Z hop as well.