Ultimaker/Cura #3116: Fusion 360 integration. Subsequent print from fusion should not start multiple Cura instances?

beikeland, 2018-01-10:
**Application Version**
3.1

**Platform**
Windows

**Steps to Reproduce**
Load/Design part in Fusion360. Select Make, 3D print,  Choose model, Check Send to 3D Print Utility, Custom (select Cura executable), Click OK.

**Actual Results**
Fusion360 prepares .stl file for model and launches Cura executable with the stl file as a parameter, and this causes Cura to load a new instance for each model printed. The new instance slows down the process. 

**Expected results**
It would be very nice if (possibly together with Fusion team) you could make Cura re-load the model, instead of starting a new instance.

**Additional Information**
[image]

ChrisTerBeke, 2018-01-10:
There's a --single-instance argument for Cura, I guess that would have to be implemented where you configure the 3D print utility application.

beikeland, 2018-01-10:
That certainly helps! Unfortunately Fusion doesn't really allow for adding this parameter, a workaround was to create a simple one line batchfile cura.bat and instead of selecting cura.exe typing cura.bat into the application selection menu in fusion. 

`@"C:\Program Files\Ultimaker Cura 3.1\cura.exe" --single-instance %* &`

Thanks for the response, I guess I'll go bug the Fusion people to try and have them add Cura to the default print applications with single instance as an option, but for now the workaround works, and can be found by others I guess.

It also seems to only work if the already running instance was started with --single-instance, that may be a feature and not a bug i guess. Changing my shortcut to Cura outside of Fusion to also include the single instance parameter should take care of that.

ChrisTerBeke, 2018-01-10:
That's intended indeed. Thanks for trying and let us know what happens with Fusion!

krisrok, 2018-02-07:
just two small improvements on @beikeland's version to make it
a) path-independent and
b) close the console window rightaway

put this into a .bat file _in the same folder_ as your cura.exe

`@start "" "%~dp0\cura.exe" --single-instance %* &`

Ohmarinus, 2020-12-02:
Hi, this solution doesn't work anymore and I'm also still having the issue where Fusion 360 opens each stl file in a new Cura instance... Would there be another way to fix this?

Latest Fusion360, Cura 4.8, Windows 10. The bat file set as target in Fusion360.

Ghostkeeper, 2020-12-04:
Indeed the --single-instance option doesn't work for me any more either. Something broke there apparently. There is no real alternative as far as I know. I'll bring this up with the team.

konskarm, 2020-12-08:
We now have an internal ticket to fix that. Devs see CURA-7916.

beardedfool, 2021-04-06:
@konskarm Is this the best ticket to subscribe to to track progress please?
Any known workarounds?
Thanks

Ghostkeeper, 2021-04-15:
@beardedfool Yes, it is. This issue is still on our backlog, and quite far down at the moment.
There is no real alternative as far as I know. No workaround. At least, short of writing a wrapper application that ensures only one instance can run at a time.

beardedfool, 2021-04-15:
@Ghostkeeper 
No worries, completely understand and thanks for coming back to me.

I can confirm that setting Preferences> "Use a single instance of Cura" does work on 4.8.0

cura.bat
```
@echo off
start "C:\Program Files\Ultimaker Cura 4.8.0\Cura.exe --single-instance" "%1"
```

closed 2021-09-15 by nallath

robbbo11, 2022-07-28:
Hi, I am new to this but found that the   @"C:\Program Files\Ultimaker Cura 3.1\cura.exe" --single-instance %* &   presented by Beikeland works for me in Win 10 and Cura Arachne beta.
@start "" "%~dp0\cura.exe" --single-instance %* &   presented by krisrok also functions as well.

commit 11b19981 (2021-09-15): Fix single instance

CURA-7916
Fixes #3116