Ultimaker/Cura #3117: Filament used reported in GUI doesn't change when filament diameter is changed

smartavionics, 2018-01-10:
**Application Version**

master branch

**Platform**

Linux

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

Load and slice model - note amount of filament used reported at the bottom and also the amount shown in the gcode file.

Change the filament diameter and reslice - note that amount reported in gui doesn't change but amount shown in gcode file changes in the correct proportion given the change in filament diameter.

**Actual Results**

GUI shows incorrect filament used amount

**Expected results**

GUI should show correct value.

**Additional Information**

ericthelin, 2018-01-18:
It is actually much worse than this issue describes. Any changes to the filament diameter are ignored! Slicing is always done using 2.85mm filament! Which makes this build completely unusable by anyone with a printer that uses 1.75mm filament! To confirm this I sliced the same model multiple times with the only change being filament diameter (changed via the per print settings and via the machine setting. The only changes in the sliced output is in commented TIME_ELAPSED and SETTINGS_3 lines

smartavionics, 2018-01-18:
Replying to the comment that has been removed - that's not true I use 1.75mm dia filament and it works OK.

ericthelin, 2018-01-18:
I removed my comment because I realized that I had inadvertently changed the gcode flavor to Marlin (volumetric). Which does ignore filament diameter because it relies on a diameter filament sensor to adjust the diameter on the fly in marlin. So my bad...

Ghostkeeper, 2018-02-05:
Can confirm. Seems to be a front-end bug.

closed 2018-04-15 by smartavionics