Ultimaker/Cura #3118: Cura 3.1.0 won't slice after update

Koentju2, 2018-01-10:
**Application Version**
3.1.0

**Platform**
Windows 10

**Display Driver**
Nvidia GeForce GT 820M

**Steps to Reproduce**
Load any model

**Actual Results**
It will start to slice and will never make any progress

**Expected results**
It slices the file and tells me how much material is required, the support and how long it'll take

**Additional Information**
Logfile included, cannot include the stl as I've signed a contract not to upload it anywhere
[cura.log](https://github.com/Ultimaker/Cura/files/1620032/cura.log)

nallath, 2018-01-17:
it seems like you have a firewall blocking the connection between the frontend and the slice engine.

ChrisTerBeke, 2018-01-17:
`UM.Backend.Backend._backendLog [93]: [Backend] [ERROR] Arcus Fatal Error (2, native 10060): Could not connect to the given address` seems to indicate that indeed.

colby2444, 2018-01-21:
I am having this issue, I just downloaded Cura 3.1. It loads slicing at .2mm but that's it. Can't go down or up, or else it crashes! This is coming up in the event log. 

Faulting application name: CuraEngine.exe, version: 0.0.0.0, time stamp: 0x5a266b31
Faulting module name: CuraEngine.exe, version: 0.0.0.0, time stamp: 0x5a266b31
Exception code: 0xc0000005
Fault offset: 0x0000000000046ee8
Faulting process id: 0x20e0
Faulting application start time: 0x01d392fd9337cd89
Faulting application path: C:\Program Files\Ultimaker Cura 3.1\CuraEngine.exe
Faulting module path: C:\Program Files\Ultimaker Cura 3.1\CuraEngine.exe
Report Id: 72942e6f-71b2-4757-bb63-e23f2dd960a1
Faulting package full name: 
Faulting package-relative application ID:

Ghostkeeper, 2018-02-08:
While your error message is different, it seems to have a similar cause. 0xc0000005 indicates an access violation. A virus scanner might be blocking it.

ianpaschal, 2018-03-27:
I'm going to consider this question answered. It's also a duplicate of a lot of others.

closed 2018-03-27 by ianpaschal

SLA-MAN, 2018-06-23:
Hi,
It is a common issue, however your conclusion raises more questions such as:
What and where is the backend and why is it required to do something that other slicers do locally?
After experiencing this same issue, and trying the suggestions such as disabling a firewall or allowing Cura traffic to the Internet, I don't understand why Internet access is required.
In my firewall settings, I can allow or disallow local network and Internet access individually.
My firewall tells me that Curaengine is trying to send info to the Internet whenever I open Cura, again whenever I load a file then once again when I try to create slice information.
When I allow Curaengine access to my local network, but block it from the Internet, why does Curaengine fail to complete slicing, but when I allow Curaengine access to the Internet, the slice process completes.
Is CuraEngine sending my loaded file and my sliced file information to your servers?

Thanks

[image]

[image]

ChrisTerBeke, 2018-06-24:
You're running Wanhou Cura. Please check with them.

Ultimaker's version of Cura slices locally, it just used 2 processes to make this more efficient.

ChrisTerBeke, 2018-06-24:
Maybe your virus scanner is misinterpreting it? CuraEngine opens a local port so it can communicate with Cura's UI. Some scanners think this needs internet, while this is not true.

SLA-MAN, 2018-06-24:
Thanks ChrisTerBeke,
After looking through the firewall history, it appears that this is the port being blocked, so I believe this is a local address.

For anyone having this concern. This appears to answer it!

SLA-MAN, 2018-06-24:
Thanks ChrisTerBeke,
After looking through the firewall history, it appears that this is the port being blocked, so I believe this is a local address.

(127.0.0.1: Port #####)

For anyone having this concern. This appears to answer it!