Ultimaker/Cura #3119: Combing Mode: Infill Only

PvtPrivate, 2018-01-11:
**Application Version**
Cura 3.1

**Platform**
Windows 10

**Steps to Reproduce**
Try to print a case with multiple screw holes inside the main body.

**Actual Results**
It prints the perimeters of the main body, then proceeds to print the hole perimeters. But when traveling from a hole to the next one, the head travels a very long distance without retracting and keeping the nozzle inside the perimeters of the main body. This results in under-extrusion on the next hole perimeter creating undersized and fragile screw holes (that are meant to be tapped).
If combing mode is set to Off, this fixes the problem but every infill line then needs a retraction. These retractions takes a lot of time and may create artifacts on the perimeter (blobs).

**Expected results**
It would be great if there is a "Infill Only" combing mode, so it will retract (and do z hop if enabled) on perimeters and skins but not on infills (or everything else?). So all the perimeters and skins are not under-extruded because of a very long combing travel.

**Additional Information**
Example print:
[image]
That particular case is not that bad, but still has the same problem. Imagine a case way bigger than that with holes that require a 5-6 seconds to travel between without retracting.

smartavionics, 2018-01-11:
Did you try setting combing to "no skin"?

PvtPrivate, 2018-01-11:
My default combing mode is "No Skin" (picture is using "No Skin"). Both "No Skin" and "All" have the same problem.

PvtPrivate, 2018-01-11:
After some searching, other people also have this problem and another suggestion is to just retract on combing. Also found this video that illustrates the problem even better: https://www.youtube.com/watch?v=4snZRGQkU_Y

smartavionics, 2018-01-11:
OK, yes, silly me, using no skin isn't going to help. Well, I can suggest two workarounds:

1 - enable "outer wall first" and "retract before outer wall" with combing enabled

2 - disable combing completely and use zig-zag infill as it will use the least travel movements.

#2 is what I do when printing runny stuff like PETG.

PvtPrivate, 2018-01-11:
I tried the option 2 you suggested, and it printed fine. But still, what if I want to print with other infill types?
Is there a "zig-zag mode" for other infills? I regularly use Cubic infill.

smartavionics, 2018-01-11:
Of all of the infill patterns printed as straight lines, zig-zag is the only one that forms chains and so is much more efficient to print (fewer travels) than the others. Is also has the advantage that it may bond to the walls better. The cross infill is made from closed polygons but it still requires travels between the polygons.

ctzsnooze, 2018-01-24:
We really need a retract on combing option, to stop blobs forming while making all the combing travels.

Ghostkeeper, 2018-02-12:
We will be implementing a zig-zag sort of technique for other infill types as well for Cura 3.3. Keep an eye on the `feature_zigzag_all_infill` branch (on CuraEngine).

Ghostkeeper, 2018-02-12:
The feature requested here is essentially to turn combing on or of when moving between infill lines separately from when moving between walls/skin (not to turn combing on or off when moving *through* infill, which is indeed the "No Skin" option).

I think that this is quite specific and a bit too opaque for a user to understand when browsing the print settings, so I'm leaning towards saying 'no' to this one.

ianpaschal, 2018-03-27:
Can reopen if @Ghostkeeper change's his mind about it; but otherwise it's going to be forgotten.

closed 2018-03-27 by ianpaschal

AbeFM, 2018-06-21:
It's not opaque, so much, as the work around for not having it is.

It sounds better to me as the default way for it to operate?

What might make sense is something that runs the retraction if the travel will be longer than some amount, but basically, I would like to turn off retraction for inter-infill moves, it runs massive number of retracts when you have gradual infill on.

Lastly: This is not new, first two hits on google were this and https://github.com/Ultimaker/Cura/issues/1901
And I was about to start my own..... If it's better gitiquette, let me know.

ctzsnooze, 2018-06-22:
I found the new retract on combing functionality to solve my concerns about combing really well.  

Mostly my problems were from not retracting when running along thin walls.  That is now so much better than before, I'm really grateful for the retract while combing option.

It's true that if the head was running over infill, I wouldn't need that retraction, because it can dribble all it likes into the infill as far as I am concerned.