Ultimaker/Cura #3120: Cura 3.2 Nozzle animation speed

Thisismydigitalself, 2018-01-11:
Will it be possible to change Nozzle's Animation Speed to match that of the print speed in mms or what ever we chose to? it seems the animation on my 3.1 is very slow and not related to print speed settings.

I would have liked to have:

- Default speed
- Manual Speed
- Actual Print speed (mms)
- Loop mode.

Thx.

jackha, 2018-01-15:
Currently every line advances at the same speed, regardless of the line length and print settings. However, you can select the "horizontal dot" next to the play button en hold the right arrow key to do a "fast forward".

Thisismydigitalself, 2018-01-15:
Thank you. it does run faster this way but what i was hoping for is to be able to run the animation fast and still be free to rotate the part freely to find where print might run into troubles.

jackha, 2018-01-16:
Such a thing would be nice, or that long lines take longer. In the meanwhile if you run from source you can change the speed in the source code ^^

In the SimulationView plugin, adjust the simulationTimer interval in SimulationView.qml

fieldOfView, 2018-01-16:
You don't need to run from source to do that, but you will need to edit a file in a text editor:
```C:\Program Files\Ultimaker Cura 3.1\plugins\SimulationView\SimulationView.qml``` (around line 650)

Thisismydigitalself, 2018-01-16:
Thank you.
I am down from 100 to 1 which is still a slow speed for my liking but much better than the default of 100. lower than 1 rendered this mode OFF. I wish to have the control over this in the GUI. to be able to speed up animation throughout the layers from bottom  to top in a loop mode so i can rotate around the part and look for troubles.

Ghostkeeper, 2018-02-12:
Our project manager removed this from our planning because it was not considered important enough.

Thisismydigitalself, 2018-02-12:
Fine with me :) it is in fact not a must have. I have lowered interval to 1 and it's good enough for when/if i animate.

clay53, 2019-09-04:
For those wanting to mess with this now, in 4.2.1, the file is now called: `SimulationViewMainComponent.qml`.

ScruffR, 2020-03-19:
I'd like to see this feature implemented too ;-)

jmattsson, 2020-09-10:
This would indeed be a nice setting to have. +1 from me. For now I guess I go looking for that qml file...

5axes, 2020-11-15:
Loop mode :

LongToeBoy, 2021-03-18:
this would be so useful feature. unfortunately 3 years passed without it...

timperlabs, 2022-04-23:
I was debugging a stringing issue on a larger model and found 3 related features that would help.

1) Playing faster as the original poster desired.
2) Showing a range of the line range as is done with the layer control.   This is possible in S3D.
3) Having a better UI element than a slider for larger models.  Even a tiny mouse movement corresponds to too large of a change in the simulation within that layer.  OTOH, the layer control is mostly fine because the layer count is not that high even with a large printer.