Ultimaker/Cura #3121: [UI Bug] Print one at a time not possible with two extruders - UM3

nicolinux, 2018-01-11:
One can not enable the setting Special Model -> Print on at a time. The setting tooltip informs the user that it is not compatible with two extruders. But it does not tell the user that it is overly cumbersome to go and edit the machine and set the extruder number to 1. This should be easier...

**Application Version**
Cura 3.1.0

**Platform**
Mac

**Steps to Reproduce**
Try to slice two objects on a UM3 and have them print one at a time.

**Actual Results**
Not possible.

**Expected results**
Possible... Cura should let me temporary set the number of extruders to 1 and then activate the print one at a time feature.

[image]

fieldOfView, 2018-01-11:
"It is not a bug, it is a feature"

This was done because Cura (generally speaking) has no idea what happens when the printer switches tools. For example in the case of the UM3 you are using, the printhead moves to a pre-defined position to switch extruders. There could be a previously printed model in the way, which could cause a collision between that print and the printhead.

That's why One at a time printing had to be disabled for multi-extrusion printers.

nicolinux, 2018-01-11:
Ok, I understand. I am not arguing about that, but I think it should be easier for the user to perform "one at a time prints" with a dual extrusion printer even if that means that the printer has to be set to one extruder.

ChrisTerBeke, 2018-01-11:
We're working on a "single extrusion mode for UM3" feature (or generally: disable extruders), which should make it easier to enable this feature from the UI.

ChrisTerBeke, 2018-01-22:
Related: https://github.com/Ultimaker/Cura/issues/1671.

yankee14, 2018-10-23:
What is the status of this? I think it was solved in #1671.

ChrisTerBeke, 2018-10-23:
It was indeed

closed 2018-10-23 by ChrisTerBeke