Ultimaker/Cura #3122: Avoid more retracts when Combing is on

MB3Y, 2018-01-11:
**Application Version**
3.1

**Platform**
Win10

**Steps to Reproduce**
MultiExtruder setup. Each object prints with other extruder. Overlap the objects in 3D space.

**Actual Results**
Retracting and travelling outside object(s) perimeters although combing is on.
Combing works fine considering each object separately, but retracts are still used when travelling from one part of object1 to the other part of object1 - despite the fact that there exists a path between the two parts of object1 where there is already printed material (or there will be, depending on the order of both extruders on this layer)
[image]

**Expected results**
When a path exists from object1.part1 to object1.part2 via objectX, then it should be preferred and used as if it was part of object1.

**Additional Information**

Ghostkeeper, 2018-01-28:
It behaves the same with single extrusion, too. The reason is that with combing we don't want to go outside the outer wall even a little, so the border that it combs within is the outer wall, but inset by half the nozzle size. After that, the shape is roughly like this (forgive me my sketching skills):
[image]

Since there are still outer walls between the two objects, they are separate as far as combing is concerned. Since you don't want combing to cut an inner corner by going all the way to the outside of the outer wall, you'd need to do some closing operation to close gaps that are the result of two adjacent objects. The result would be roughly like this then:
[image]

This requires two offsets extra on all layers. I think this is going to be too expensive computing-wise for what you gain from it. Any other opinions?

MB3Y, 2018-02-08:
_I hope I have some terminolgy right..._
In the above solution you suggest to calculate the boundaries for the "new" combing by combining the "existing" combing contours of the different objects and do some "gap closing".
What if you first merge all objects into a virtual object/mesh/stl and you calculate the "new" combing contours based on that merged object? In layers where you have completely separated areas it'll work like it does already today - in layers where the different objects appear "merged" the combing contour would be as suggested.
Would such approch be possible, save time or be simpler?

Ghostkeeper, 2018-03-02:
> What if you first merge all objects into a virtual object/mesh/stl

I think you're suggesting to combine the two 3D models, and use the combined mesh for planning the combing move. Combining two 3D models is a much harder problem though. Algorithms to do that are few and also take a lot of computation time.