Ultimaker/Cura #3126: Option to edit printer settings missing?

smartavionics, 2018-01-12:
**Application Version**

master branch

**Platform**

Linux

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

Open manage printers dialog.

**Actual Results**

Dialog opens but there is no button to edit the printer settings.

**Expected results**

I expected a button to be visible that would open the machine settings dialog.

BTW - removing the printer and adding it again makes no difference.

**Additional Information**

[image]

fieldOfView, 2018-01-12:
I can't reproduce this on current master. Logs?

Looks like some plugins fail to load (MachineSettingsAction, and OctoPrintPlugin too?)

smartavionics, 2018-01-12:
Oops! My bad, somehow, that plugin had been disabled. Must have been finger trouble on my part. Sorry to waste your time. Closing...

closed 2018-01-12 by smartavionics