Ultimaker/Cura #3127: Layer view

vmzsoft, 2018-01-12:
**Application Version**

Ultimaker Cura 3.0.1
**Platform**

Windows 7
**Qt**

**PyQt**

**Display Driver**

NVIDIA PhisX 32 bits
**Steps to Reproduce**

-
**Actual Results**

-
**Expected results**

-
**Additional Information**

Very bad Layer view. Why you don't want to make the same beautiful and pleasing to the eye such as in Slic3r?

vmzsoft, 2018-01-12:
Very bad "Layer view". Why you don't want to make the same beautiful and pleasing to the eye such as in Slic3r?

smartavionics, 2018-01-12:
Please include a screen shot so we can see why it is bad, thanks.

vmzsoft, 2018-01-13:
Re: [Ultimaker/Cura] Layer view (#3127)

Hi, Mark.

You wrote 12 января 2018 г., 22:40:43:

Please include a screen shot so we can see why it is bad, thanks.
—

Nothing could be simpler.
Please here are two images of the same part.
Where do you see a more detailed image without the risk of ruin your eyes?

Best regards,
VMZ                       mailto:vmzsoft@gmail.com

fieldOfView, 2018-01-13:
> Where do you see a more detailed image without the risk of ruin your eyes?

We don't see anything at all, because github does not handle attachments in messages.

Please use your words to describe what is better about the layer view in Slic3r, and what you want to see changed in Cura. If you just say "make it more pleasing" without being more specific, chances anything is going to change because of that are very low.

smartavionics, 2018-01-13:
You can upload an image to github using the instructions given below this text box, like this...
[image]

vmzsoft, 2018-01-13:
Re: [Ultimaker/Cura] Layer view (#3127)

Hi, Aldo.

You wrote 13 января 2018 г., 12:25:17:

Where do you see a more detailed image without the risk of ruin your eyes?

Did you ever try to contact an eye doctor?

-- 
Best regards,
VMZ                       mailto:vmzsoft@gmail.com

vmzsoft, 2018-01-13:
Re: [Ultimaker/Cura] Layer view (#3127)

Hi, Mark.

You wrote 13 января 2018 г., 12:37:07:

You can upload an image to github using the instructions given below this text box, like this...

For what? To a lot of people agreed with me?
I suggest you make Cura is even better.
To Cura became a leader and a favorite program for all.
Your programmers can do high-level view layers very nicely.
The envy of all.
It is strange that not all support it.
Your comrade Aldo Hoeben generally stated that he sees no difference in the images.
I advised him to consult an optometrist.

-- 
Best regards,
VMZ                       mailto:vmzsoft@gmail.com

smartavionics, 2018-01-13:
Haha, I get it, you're a troll. Goodbye!

vmzsoft, 2018-01-13:
Re: [Ultimaker/Cura] Layer view (#3127)

Hi, Mark.

You wrote 13 января 2018 г., 16:28:25:

Haha, I get it, you're a troll. Goodbye!
—

Your logic is very severely affected.
I'm not a troll, but a real person who wishes you well.
But you somehow struggling trying to seem stupider than you really are.
Yes, fools on the light a lot.
But I did not expect that their gathering place is at your address.
It is very unfortunate.
Well, in that case I remain adherent to Slic3r ...
More I will not bother you.
Because you like to be losers...
Adju, stupid Amigo ...

Best regards,
VMZ                       mailto:vmzsoft@gmail.com

vmzsoft, 2018-01-13:
Pictures:
http://dropmefiles.com/VJBhr

smartavionics, 2018-01-13:
That's better, we can see what you are talking about now. Cura is using display compatibility mode which means that you have either forced compatibility by using the preference "force layer view compatibility mode" or your computer does not support the more recent opengl version that Cura would like. I hope this helps.

vmzsoft, 2018-01-13:
Re: [Ultimaker/Cura] Layer view (#3127)

Hi, Mark.

You wrote 13 января 2018 г., 20:32:19:

That's better, we can see what you are talking about now. Cura is using display compatibility mode which means that you have either forced compatibility by using the preference "force layer view compatibility mode" or your computer does not support the more recent opengl version that Cura would like. I hope this helps.
—

Thank you for your answer.
Would you recommend me to buy another computer?
I myself am a programmer (mostly on databases), but I would never instruct users to what they need a computer to use my program.
Or you it seems strange?

Best regards,
VMZ                       mailto:vmzsoft@gmail.com

smartavionics, 2018-01-13:
I was faced with a similar situation a while back and the only solution to get the better layer view was to buy new hardware. I know nothing about your computer but perhaps it is sufficient if you update your graphics drivers to the latest versions?

ChrisTerBeke, 2018-01-14:
Not a Cura issue as compatibility mode is either activated here in preferences or the hardware doesn't handle normal layer view, closing.

closed 2018-01-14 by ChrisTerBeke

jackha, 2018-01-15:
Cura requires at least OpenGL 4.1 for the "new" layer view.