Ultimaker/Cura #3128: Suggestion for 3.2/3: Nozzle Animation to include START/END Gcode as well

Thisismydigitalself, 2018-01-12:
Please allow us to include START/END Commands to Nozzle Animation.

jyc66, 2018-01-13:
This seems like a wise idea

Ghostkeeper, 2018-01-28:
This has some difficulties. We're not displaying the start/end g-code in layer view at all, at the moment. The reason for this is that we pre-generate both sides of every line (with some line width) in the slicing engine. The engine knows these coordinates after all the insets it's already done for slicing, so to speed up the generation of layer view they are sent back to the front-end in the form of polygons that it can show in layer view.

It doesn't do this for the start and end g-code. Those are not generated by the engine after all. You'd have to generate them. We currently do this when you load a g-code file into Cura, which then takes two orders of magnitude more time to process. To have the speed of the CuraEngine-generated lines but the completeness of the front-end generated lines, we'd have to sort of make a mix of these two techniques but that messes with the data model in the front-end. Let's just say it's complex.

What's more, the start g-code often has no moves at all, and if it has one it is often a vertical move too which we wouldn't be able to display well. You can still see the start and end g-code if you save the g-code to file and then load it back into Cura.

Thisismydigitalself, 2018-01-30:
Thank you for your elaborate explanation.
What made me ask for this feature was following me fiddling with my new printer's Start code in trying to make the nozzle go to a specific position and extrude a given amount of filament before printing.
for END code i wanted to push the bed to a specific location for best printed part's photo-up by the printer's webcam. i managed to smash the nozzle to bed few times till i managed to get the right gcode commands. having animation for START/END stages would have eliminated all of that.

Ghostkeeper, 2018-03-26:
Our project manager removed this from our planning. It's deferred for now.