Ultimaker/Cura #3130: Cura 3.1 MacOs 10.13 "Unable to slice" when retraction is enabled

zekitez, 2018-01-14:
Cura 3.1
MacOs 10.13.1

Enabling retraction in Cura 3.1 causes an error message "Unable to Slice. The following settings have errors: Retraction Retract speed, Retraction Prime Speed, Retraction Speed."
This message does not state what the error is.
Disabling retraction does not solve the error, it remains.
Either the error message is wrong and something else is wrong or retraction... 

The same settings works perfectly in version 2.6.2 .
The screenshot has Cura 2.6.2 on top and Cura 3.1.

[image]

jackha, 2018-01-15:
Could you provide us with more info? Either attach a workspace that reproduces this problem or provide reproduction steps. Thanks :-)

zekitez, 2018-01-16:
Attached the zipped configuration folder.
Unzip it and replace it with your 3.1 configuration folder.
Start Cura 3.1.
The selected printer, with the error, is named MyPrinterFail.
Open any stl file.
Within several seconds or after changing a parameter the "Unable to Slice. The following settings have errors: Retraction Retract speed, Retraction Prime Speed, Retraction Speed." is shown.
If you now change to another printer (Creality CR-10) then slicing will be succesful.
I have created a "MyPrinter #2" with the same parameters as MyPrinterFail and slicing for "MyPrinter #2" is succesful. I assume that within MyPrinterFail something is corrupted.

[3.1.zip](https://github.com/Ultimaker/Cura/files/1636260/3.1.zip)

Ghostkeeper, 2018-01-30:
The retraction speed settings give an error when the retraction speed is 0 or lower, or when the retraction speed is higher than the maximum E feedrate of the printer. The maximum E feedrate of Custom FDM Printer is 299792458000 mm/s.

I see nothing wrong in your configuration.

But when I test your configuration on my computer and inspect the settings in memory I see that it evaluates these three retraction settings as "0" on all inactive stacks. Under water, the Custom FDM Printer always has 8 stacks because it was difficult to actually have a variable number of stacks. It just doesn't display all of them.

This was fixed in 3.2. But a workaround for 3.1 would be to say that your printer has 8 extruders (just not use them).

Ghostkeeper, 2018-01-30:
Actually, looking a bit deeper into it, it seems that you've added a custom material called "Transparent", which sets the retraction speed to 0. This is not allowed. You should be able to edit that material and set the retraction speed to something else.

ianpaschal, 2018-03-26:
> This was fixed in 3.2.

Also re: "Transparent", changing that should also fix it.

So, closing.

closed 2018-03-26 by ianpaschal