Ultimaker/Cura #3131: Cura 3.1 MacOs: Inconsistent use of decimal seperator

zekitez, 2018-01-14:
Cura 3.1
Macos 10.13

Inconsistent use of decimal seperator. 

Mostly the dot is used as seperator but sometimes a comma is used, at least the fields:
- Preferences -> Materials -> Information    : Properties Density    uses a comma
- Preferences -> Materials -> Information    : Properties Diameter   uses a comma
- Preferences -> Materials -> Information    : Properties Filament Cost   uses a comma
- Preferences -> Materials -> Print Settings : Retraction Distance   uses a comma

ChrisTerBeke, 2018-01-17:
That should indeed be the same in all places.

ChrisTerBeke, 2018-01-17:
Devs: CURA-4822

Ghostkeeper, 2018-02-05:
Those fields use the proper locale of your system (because they are system-generated). Cura does not and just always uses a period.

I looked into this and found one solution that's available to us in QML: `toLocaleString`. However this function always outputs a fixed number of decimals, so `3` becomes `3,00`, `3.1` becomes `3,10` and `3.14159` becomes `3,14`. To fix this we'd have to write a custom function and put it in every QML file. We're not about to do this, so unless we find a better solution from QML we'll leave this bug in.

closed 2018-08-08 by ianpaschal