Ultimaker/Cura #3132: Can't edit Start/End Gcode per extruder

MB3Y, 2018-01-14:
**Application Version**
3.1

**Platform**
Win10

**Steps to Reproduce**
I have a custom FDM printer created with 3 extruders.
Goto Machine settings, select the tab for Extruder 1, 2 or 3, add some Start Gcode and press the Close button.

**Actual Results**
Edits in Extruders tabs are not saved. Start/End Gcode boxes are empty again - also gcode does not contain the edits done before.

Thisismydigitalself, 2018-01-15:
Two days ago I noticed the same thing for a single extruder printer. Can't remember how i fixed that but i think saving a project saved the codes as well but before saving each time i closed the tab/window and reopened start/end tab my settings reverted to the ones before closing the tab.
Cura 3.1R.

ChrisTerBeke, 2018-01-17:
@LipuFei also per-extruder settings?

LipuFei, 2018-01-17:
@ChrisTerBeke Yes, together with that definition_changes bug. It's fixed in 3.2.

closed 2018-01-17 by ChrisTerBeke

Thisismydigitalself, 2018-01-20:
Just happened to me again. annoying.
Good to know this is solved in 3.2.