Ultimaker/Cura #3133: Feature Suggestion: Keep Selected Post-Processing rules in place after shutdown and restart of Cura

MB3Y, 2018-01-14:
**Application Version**
3.1
**Platform**
Win10

**Steps to Reproduce**
Set some Pros-Processing rules.
Use them in the selected model.
Notice these rules keep active, even when clearing the buildplate and selecting a new model.
Shutdown Cura.
Restart Cura.

**Actual Results**
Post-Processing rules are not active anymore.
If you want to use them again, you have to create them again.

**Expected results**
Make Post-Processing rules persistant over restarts of Cura.
As a flagable  option maybe in the Post-Processor itself.

**Additional Information**
Idea came up since I use the post-processor to compensate/as workaround for printer mis-interpretations of standard GCode. As such I should keep them rules active for any slicejob. See also #2918.

MB3Y, 2018-01-15:
> This is done by design currently as more people found it annoying that the scripts were still there and they forgot about it, screwing up the new prints.

Wouldn't it be possible to keep all happy by letting people the choice of keeping the settings permanent or not? Intermediate step could also be to store the formerly used rules such that you can select previous rules instead of forming them from scratch again?

alekseisasin, 2018-01-17:
Actually what you are asking is to store Post-Processing settings. To save them to external location, I think it is doable, only to support for different versions might be tricky. (I mean if you save in cura 3.1 then it will work for 3.1 without guarantee for the next version)

ChrisTerBeke, 2018-01-17:
Duplicate of #2796 and #2854.

closed 2018-01-17 by ChrisTerBeke