Ultimaker/Cura #3135: 3.1.0 Crashing

TectG34, 2018-01-15:
**Application Version**
3.1.0

**Platform**
Window 10 

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

Cura 2.6 works ok till I upgraded to Windows 10. There a small issue with the opening files in 2.6.0. The first time you used the program, you had to open the file dialog box, close it and reopen it, to get it to work. It worked fine after that.
I them upgraded to Windows 10 and Cura 3.1.0 and now when I try to open a file, to load a model, the program freezes and then crashes. 

Tried to go back to 3.0.0 and it did not work.
I tried 2.6.0 and it did not work.
Reinstalled Windows 10 from a USB key, Cura 3.1.0  still crashes when opening a file. 
Started Cura, run as admin, still crashes

Any Idea why Cura crashes when trying to open a file?

**Actual Results**
Still crashes

**Expected results**
That Cura would go back to being a great program :-(

**Additional Information**

ChrisTerBeke, 2018-01-15:
Can you attach Cura logs? Then we can figure out what's going on.

TectG34, 2018-01-16:
Thanks Chris for getting back to me. Would you believe I booted the program this morning to make sure the log would have a crash report on it and Cura worked perfectly. It seem that your mere reply frightened the program into working.

I did do one other thing this morning, I finally got rid of a Dell recovery software that kept popping up and asking if I wanted set up their recovery software. I do not know if this was causing a conflict, but I uninstalled it and the problem is gone. 

Despite this fact, I like to think you reply frightened the program into working. Thanks again for getting back to me.

closed 2018-01-16 by ChrisTerBeke

ajb142, 2018-08-01:
I know this is closed, but i thought this might be useful. 

I had this same problem and fixed it by uninstalling the dell backup and recovery software.