Ultimaker/Cura #3136: Calibration

azeemiqbal123, 2018-01-16:
Is there calibration option built in Cura? It would be great that a simple calibration script may be available to allow swift bed level calibration. Further, an option to start/stop extruder should be there independently in manual control. This helps greatly when the nozzle is clogged. 

**Application Version**

**Platform**

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

ChrisTerBeke, 2018-01-16:
It's not, and probably won't be. The UM3 (our main focus) has auto-calibration. 3rd party printers cannot be supported by us as there are so many out there.

closed 2018-01-16 by ChrisTerBeke

fieldOfView, 2018-01-16:
There is a somewhat rudimentary buildplate calibration option for the Ultimaker Original that can be "turned on" for other printers, but you would need to edit the definition file for that. It could be fleshed out into a more functional plugin for more printers (that could also be compatible with the OctoPrintPlugin).

Ghostkeeper, 2018-01-30:
I think the auto-calibration sequence for the UMO should work for most printers out there (as long as they are connected via USB). It uses the width and height of the build volume, but adds some constants there too so the build plate should at least be 20mm wide and high, but we can assume that all printers are at least that big.