Ultimaker/Cura #3138: Adding new printer

maukcc, 2018-01-16:
We want to add our new Cartesio434 machine to the definitions list. 
Is it absolutely necessary to also add new : 
- extruders
- materials
- quality profiles
- variants
as they are all exactly the same as those for the old Cartesio machine.
Or is it possible to also link these to the new machine.

I really do not want to make the definitions list any longer(as it will take longer to startup cura), but we need to, as we have other disallowed areas.

ChrisTerBeke, 2018-01-16:
You can use the inherited field to base a printer definition off of an existing one. Check the Ultimaker definitions for examples.

fieldOfView, 2018-01-16:
The "inherits" field only helps with the definition, not with the extruders, materials, qualities and variants.

Have a look at the Ultimaker 3 Extended. It uses the "inherits" field, but it also specifies the "quality_definition" to point to the quality profiles of the Ultimaker 3.

maukcc, 2018-01-18:
I can not see any changes happening with the "inherits" changed. What should that do?

When adding "quality_definition": "cartesio" 
it actually also changes the variants and materials. I am happy with that :)

In the extruder files we did:

>  "metadata": {
>         "machine": "cartesio",
>         "machine": "cartesio434",
>         "position": "0"
> },

This way I did not need to make 4 extra extruder files. It may be a good thing to do this for the other machines as well, to get rid of redundant files.

So we actually only needed to make 1 more definitions file :)
Which is still a shame as it will slow down startup, and is only needed for the mesh and disallowed area.

maukcc, 2018-01-18:
The Extruder thing did not work as it breaks the link to the machine that is mentioned 1st.

thetobby, 2018-01-18:
I had the same question since all the printers seems to be setup differently since there are different authors. You do not need materials. Check out my Anycubic i3 for reference. It only uses Definition, mesh(optional) and quality presets.

ChrisTerBeke, 2018-01-22:
Do you need custom extruder definitions? That's only needed if they are different from a regular FDM extruder (e.g. with dual extrusion setups where both extruders have different offsets).

maukcc, 2018-01-23:
Yes, we have up to 4 extruders(or other tools), all with offsets and start and end gcodes.

ChrisTerBeke, 2018-01-23:
Are those extruders the same for all printers? Or at least share some properties?

maukcc, 2018-01-23:
Yes they are the same.

ChrisTerBeke, 2018-01-23:
Then you could use inheritance for those shared properties (e.g. make a base extruder definition and machine definition) and then make separate definitions for the machine and extruders that the user should see in the interface and inherit from those base ones.

cc @LipuFei is this correct?

ChrisTerBeke, 2018-01-23:
@fieldOfView that's correct, but since they're different printers I guess some params are different anyways between those extruders.

maukcc, 2018-01-23:
No, they are exactly the same. That is why we think it is silly to have them in there twice (or 3x or 4x or .....)

ChrisTerBeke, 2018-01-23:
Can't you reference those extruders in the different machines? We do the same with UM3 and UM3 Extended.

maukcc, 2018-01-23:
Every multi extruder definition file has their own extruder files, even if the extruders are exactly the same, like the UM3 and UM3e.
The point of this issue is that this generates unnecessary files(10 at present, 14 if we add the Cartesio extruders) and space.

maukcc, 2018-01-23:
The way I would like to see the definitions is:
There is only 1 definition file for Cartesio.
When the user chooses Cartesio from the "other" printers list, they get to machine settings, and in machine settings they choose which Cartesio they have (4 choises). 
You can change the Cartesio name with UM if you like that more :)

fieldOfView, 2018-01-23:
> Can't you reference those extruders in the different machines? 

No, because extruders have a backreference to the machine.

> We do the same with UM3 and UM3 Extended.

No you don't. UM3 Extended has its own set of extruders.

ChrisTerBeke, 2018-01-23:
Then I'm afraid separate files are needed. The quality definition can be shared though.

closed 2018-02-09 by maukcc