Ultimaker/Cura #3139: Import .xml material file with missing tag in the XML cause material cannot be deleted anymore

noisemaker00, 2018-01-16:
Import a bad formatted material XML (eg. without GUID, etc) cause the material will be added to the list but it cannot be removed anymore (where those XML are saved to remove them manually??)
I suggest to add a syntax check on the XML to avoid potential issues.

alekseisasin, 2018-01-17:
What OS system do you use ?

noisemaker00, 2018-01-17:
Windows 7 x64

ChrisTerBeke, 2018-01-22:
You can remove the custom material from your user data folder (https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations) for now. Of course loading an incorrect XML file shouldn't totally break your Cura install, so we'll put that on our internal ticket system.

ChrisTerBeke, 2018-01-22:
Devs: CURA-4851

noisemaker00, 2018-01-25:
In addiction to this, I had problem even to editing a material without a name.
I had to open the material XML file manually and editing the `<label>` tag

alekseisasin, 2018-01-31:
@noisemaker00  What do you expect from us to do with <label> tag

noisemaker00, 2018-01-31:
I don't know how is handled internally, but I'm a software developer too and if a field is missing I think an exception handler should be implemented. Is quite frustrating when you add a material without name to be unable to open it again from the materials manager screen.
No all materials have names, because they are easily identified using only manufacturer/material type and color

alekseisasin, 2018-01-31:
So far as I know the label tag in not required, it is optional. If labels is missing the name was defined different. 
For now I am checking the implementation and there will be some changes.
Do you use the latest Cura 3.1 version ?

LipuFei, 2018-01-31:
Hi @noisemaker00 , a fix has just been made to validate an imported material file as you have suggested:
- https://github.com/Ultimaker/Cura/commit/e5c72cfdc5cef3d1cb995d5e9504e45fb94216c6
- https://github.com/Ultimaker/Cura/commit/a446ca2759170b78bb17d14fdc15c9d5c9e00010

Ghostkeeper, 2018-02-26:
This issue has been verified by QA and should be fixed in Cura 3.3.

closed 2018-02-26 by Ghostkeeper