Ultimaker/Cura #3140: Material link UI issues

peteruithoven, 2018-01-16:
**Application Version**
Cura 3.1.0

When working with the Ultimaker 3 you have to print with a material that's linked to the material type selected on the printer. Otherwise you'll get a message on the printer that you have to change this. (You can only select Ultimaker's material types.)
But when creating a new material from scratch it's not linked, or it's linked to PLA. You'll have to duplicate a temperature compatible Ultimaker material type. Which seems like the only way to create a link, which is quite unintuitive. 

Could an add / edit linked material type be added? 

To which material is this material linked?
[image]

Can't edit material link, I can only remove it. 
[image]
[image]

ChrisTerBeke, 2018-01-17:
The linking is supposed to work like this (duplicating a pre-shipped material results in a link, which can be removed to allow editing all fields independently). So I don't think it's gonna change?

cc @Appesteijn

peteruithoven, 2018-01-17:
> The linking is supposed to work like this

But why? Why should I not be able to add / edit a link afterwards?

Is there anything else in the pipeline to make the use of custom materials on an Ultimaker 3 smoother?

ChrisTerBeke, 2018-01-22:
I don't know how or when this was decided, only that it's the current behaviour. I think the idea was that if you unlink you're creating a totally custom material and re-linking after that can give a lot of issues with regards to upgrading and backwards compatibility.

I'm not sure about custom material on Ultimaker 3, but I don't have any issues using one of the generic materials and sending jobs to the UM3 over the network. Next version of Cura Connect will be a bit more generous with scheduling jobs when the material has a mismatch of brands (but is still the same type).

Ghostkeeper, 2018-02-08:
We had it like this for the sake of simplicity. Adding an interface to re-link materials to other materials is complex and for most people unnecessary. If you only need the quality profiles, simply setting the material type should be enough.

You should be able to print on an UM3 with your custom material though. You just get a warning message on the printer that it is not advisable to print with that material (which is very annoying, I grant you).

peteruithoven, 2018-02-20:
> You just get a warning message on the printer that it is not advisable to print with that material 

That's indeed the biggest issue. It's hard to imagine what it's impact is, how the selected filament affects the printer. I've mostly assumed that it only affects things like during switching filament though the menu's, what temperature is used. But there really is nothing that gives that kind of information.

Ghostkeeper, 2018-03-12:
The effect is that the printer doesn't know the material type to use. It uses this material type in order to do the bed levelling at the correct temperature, priming after a material switch, what temperature to heat for removing the material, etc. You can tell the printer which material profile to use from its own list, and you can choose one that comes close in the temperature it needs to use (PLA for cold-ish materials and PC for hot-ish materials is what I usually do).

Anyway, I don't think that re-linking materials manually is desired. It would add too much complexity to the interface for too little gain, in my opinion.