Ultimaker/Cura #3141: [FR] Slicing / Output Resolution Setting

CCS86, 2018-01-16:
Has this been considered before? 

High resolution STL files can generate gcode that is incredibly dense, creating commanded moves smaller than a single microstep. I believe Slic3r has a similar setting to limit slicing resolution.

ChrisTerBeke, 2018-01-16:
This already exists in Cura. By default it's 10 microns, but you can configure it in settings.

closed 2018-01-16 by ChrisTerBeke

CCS86, 2018-01-16:
Sorry!

CCS86, 2018-01-18:
@ChrisTerBeke, do you know who might know the details of this setting?

Using an STL file output with a resolution of 0.005mm, and a Cura setting of 0.010mm, the line segments output, in very small radii, are about 0.155mm. 

Loosening the Cura setting tends to induce random error into the slicing. where a vertical walled part, has inconsistent path placement, layer to layer:

[image]

[image]

ChrisTerBeke, 2018-01-18:
That's expected as each layer's polygons can be slightly different due to where the layer intersects the mesh triangles. It's not meant as a way to add 'low poly' printing, and usually you don't want a value higher than 10-15 microns.

CCS86, 2018-01-19:
Yeah, that's what I am seeing. This setting seems to have a finite range that gives good results. I am fighting with Marlin motion issues, when it starts choking on too many segments/s (https://github.com/MarlinFirmware/Marlin/issues/9219)

With my own designs, I can control this with output resolution. But downloaded models are sometimes very dense, and I'm trying to find a good way to control this. Was hoping Cura could simulate lowering the model resolution, but that's a tall order.

Ghostkeeper, 2018-02-06:
The resolution limit was always there but recently it's been made into a setting.

It lowers the resolution per-polygon by going through the coordinates of the polygon and removing every coordinate that is closer to the previous coordinate than this resolution setting. There is no assurance that the coordinates that are removed are the same coordinates as were removed on the previous layer. That causes the irregularities. This is very hard to assure after slicing, since there is no guarantee that the coordinates are in the same position across layers or that there is an equal number of coordinates. Or even that the polygon exists there. Maybe two polygons merged together on that layer. And so on.

You'd have to do this in 3D by removing to do it properly. I usually take Blender's decimate filter which works really well I think. It's probably not going to be implemented in CuraEngine because algorithms on 3D space are expensive to compute and hard to implement reliably.

CCS86, 2018-02-06:
Thanks @Ghostkeeper 

I can see how it would be very hard to see a consistent result with that method. Even vertical walls have facets with random orientation, so during slicing the line segments they create start and stop in random places.

I tried decimate in Blender, but found that it destroyed the junction between a planar top/bottom surface of a model, and the more complex sides. 

For my own work, it's easy to output at an appropriate resolution. But downloaded stuff often is all wrong.

emericrobert, 2018-11-23:
Hi, I do have a similar problem, and can't find a solution... yet!

The STL have no small detail (it's an egg shape)
It was generated from SolidWorks with custom export setting to increase STL resolution to 0.05mm
the STL look very smooth in Cura preview (same in other STL viewer), but facets are visible after slicing in the layer view mode.
I try activating or desactivating ''smoothing'' feature in cura like the ''smooth spiralized contours'' with no visible effect. I also tried to play with the ''maximum resolution'' setting at value around 0.1mm

of course, the facets are present in the printed part. the facette are around 1cm in size after printing

Other info:
I'm slicing with cura 3.6.0
Printing with a CR10S with 1mm nozzle using ''spiralize outer contour'' 

files (Gcode and STL) are availible here if anyone want to take a look:
https://drive.google.com/drive/folders/1UjThFmO2lnqGOdLvkIqhEtqQYbnosOnR?usp=sharing

[image]

[image]