Ultimaker/Cura #3143: Nozzle size set in definitions for single extrusion printers not functional

fieldOfView, 2018-01-17:
**Application Version**
3.2

**Platform**
any

**Steps to Reproduce**
Add a new printer of type Deltabot, Bfb, 3deator, Grr neo, and possibly others
Go into Machine Settings to check the nozzle size

**Actual Results**
The nozzle size for the single extruder is set to 0.4

**Expected results**
All mentioned definitions specify the machine_nozzle_size as 0.5

**Additional Information**
This is likely caused by the rework of global-only printers into global+extruder stack.

ChrisTerBeke, 2018-01-17:
@LipuFei sounds related to the per-extruder settings issues we had?

ChrisTerBeke, 2018-01-25:
@LipuFei ping :)

Ghostkeeper, 2018-02-13:
Seems like this is fixed now. I can't reproduce this in the Master branch. My steps:

1. Add Deltabot.
2. In Machine Settings at Extruder 1, check the nozzle size. This is properly 0.5mm.
3. Check the line width. This is also properly 0.5mm.

closed 2018-02-13 by Ghostkeeper