Ultimaker/Cura #3145: UM3 not printing on Extruder 2 - issue began after 3.1.0 update

DanEils, 2018-01-17:
**Application Version**
3.1.0

**Platform**
Mac OS High Sierra

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**
Extruder 1 - AA 0.8 PC
Extruder 2 - AA 0.4 PLA

Try printing with Extruder 2 -  right click - choose extruder 2 for the part, try to print, 

**Actual Results**
No matter what I do, even after restarting the printer and computer, I cannot make the UM3 print with Extruder 2 only as I was easily able to do before the 3.1.0 update. Yes, I have selected the part to print on Extruder 2 by right clicking on the part and choosing the extruder.

**Expected results**
I would like to be able to print on extruder 2 only with my AA 0.4 nozzle

**Additional Information**

ChrisTerBeke, 2018-01-17:
Are you able to slice?

DanEils, 2018-01-17:
I apologize for not providing more detail,

Yes, curiously it slices based on the extrusion width setting for the
extruder 2, the file transfers and it goes through the preprint leveling as
usual, then it keeps starting the print on extruder 1 instead of extruder 2.

This is not a high priority for me. I'm not printing multi material so I
finally gave up trouble shooting and just manually switched out my Extruder
1 material from PC to PLA for this print and resliced for the wider nozzle.

ChrisTerBeke, 2018-01-17:
It might be that the skirt or brim is still printed with extruder 1? You can check the layer view and set the color mode to 'material color' to see the exact slicing results before printing.

DanEils, 2018-01-17:
I think I checked the skirt extruder, but I can't say for sure. I will
double check after this print finishes and let you know.

I hope this is it! Thanks.

Ghostkeeper, 2018-02-05:
That's a pretty long print then :wink:

ianpaschal, 2018-03-26:
Since we didn't hear back from OP and this should be a non-issue in 3.3 with single-extrusion mode, I'm closing this.

closed 2018-03-26 by ianpaschal

DanEils, 2018-06-12:
Yes, it was simply the default extruder for the skirt. Sorry for wasting any time. It was a long print :)