Ultimaker/Cura #3146: Layer view crash

BagelOrb, 2018-01-17:
**Application Version**
3.1.0 AppImage 

**Platform**Linux Mint (Ubuntu) 18.2 Cinnamon

**Display Driver**
Nvidia xserver-xorg-video-nouveau 1:1.0.12-1build2

**Steps to Reproduce**
Load attached project or gcode

**Actual Results**
Cura suddenly shuts down (crash)
Nothing in logs.

**Expected results**
No crash

[hatch_even_3.curaproject.3mf.txt](https://github.com/Ultimaker/Cura/files/1640204/hatch_even_3.curaproject.3mf.txt)
[UM3_hatch_even_3_crash.gcode.7z.txt](https://github.com/Ultimaker/Cura/files/1640210/UM3_hatch_even_3_crash.gcode.7z.txt)

smartavionics, 2018-01-17:
Hi Tim, I loaded the gcode  into master branch Cura (took about 5mins!) and then it blew up:

````
2018-01-17 23:11:10,162 - DEBUG - GCodeReader.FlavorParser.processGCodeFile [444]: Finished parsing /home/markb/Downloads/UM3_hatch_even_3_crash.gcode
QObject::killTimer: Timers cannot be stopped from another thread
2018-01-17 23:11:10,172 - DEBUG - GCodeReader.FlavorParser.processGCodeFile [457]: Loaded /home/markb/Downloads/UM3_hatch_even_3_crash.gcode
2018-01-17 23:11:10,172 - DEBUG - UM.FileHandler.ReadFileJob.run [68]: Loading file took 242.8 seconds
2018-01-17 23:11:10,175 - WARNING - UM.Qt.QtApplication.__onQmlWarning [269]: file:///opt/cura/lib/qml/QtQuick/Controls/TextField.qml:638:5: QML TextInputWithHandles: Binding loop detected for property "text"
2018-01-17 23:11:10,215 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught error has occurred!
2018-01-17 23:11:10,219 - CRITICAL - cura.CrashHandler.__init__ [65]: Traceback (most recent call last):
2018-01-17 23:11:10,220 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/cura/CuraApplication.py", line 789, in event
2018-01-17 23:11:10,220 - CRITICAL - cura.CrashHandler.__init__ [65]:     return super().event(event)
2018-01-17 23:11:10,220 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 341, in event
2018-01-17 23:11:10,220 - CRITICAL - cura.CrashHandler.__init__ [65]:     event._function_event.call()
2018-01-17 23:11:10,221 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/UM/Event.py", line 189, in call
2018-01-17 23:11:10,221 - CRITICAL - cura.CrashHandler.__init__ [65]:     self._function(*self._args, **self._kwargs)
2018-01-17 23:11:10,221 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/UM/Signal.py", line 315, in __performEmit
2018-01-17 23:11:10,222 - CRITICAL - cura.CrashHandler.__init__ [65]:     func(dest, *args, **kwargs)
2018-01-17 23:11:10,222 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/cura/CuraApplication.py", line 1508, in _readMeshFinished
2018-01-17 23:11:10,222 - CRITICAL - cura.CrashHandler.__init__ [65]:     op.push()
2018-01-17 23:11:10,222 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/UM/Operations/Operation.py", line 50, in push
2018-01-17 23:11:10,223 - CRITICAL - cura.CrashHandler.__init__ [65]:     Application.getInstance().getOperationStack().push(self)
2018-01-17 23:11:10,223 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/UM/Operations/OperationStack.py", line 52, in push
2018-01-17 23:11:10,223 - CRITICAL - cura.CrashHandler.__init__ [65]:     operation.redo()
2018-01-17 23:11:10,223 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/UM/Operations/AddSceneNodeOperation.py", line 40, in redo
2018-01-17 23:11:10,224 - CRITICAL - cura.CrashHandler.__init__ [65]:     self._node.setParent(self._parent)
2018-01-17 23:11:10,224 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/UM/Scene/SceneNode.py", line 274, in setParent
2018-01-17 23:11:10,224 - CRITICAL - cura.CrashHandler.__init__ [65]:     scene_node.addChild(self)
2018-01-17 23:11:10,224 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/UM/Scene/SceneNode.py", line 357, in addChild
2018-01-17 23:11:10,225 - CRITICAL - cura.CrashHandler.__init__ [65]:     self.childrenChanged.emit(self)
2018-01-17 23:11:10,225 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/UM/Signal.py", line 212, in emit
2018-01-17 23:11:10,225 - CRITICAL - cura.CrashHandler.__init__ [65]:     self.__performEmit(*args, **kwargs)
2018-01-17 23:11:10,225 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/UM/Signal.py", line 315, in __performEmit
2018-01-17 23:11:10,226 - CRITICAL - cura.CrashHandler.__init__ [65]:     func(dest, *args, **kwargs)
2018-01-17 23:11:10,226 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/cura/plugins/SimulationView/SimulationView.py", line 159, in _onSceneChanged
2018-01-17 23:11:10,226 - CRITICAL - cura.CrashHandler.__init__ [65]:     self.calculateMaxLayers()
2018-01-17 23:11:10,226 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/cura/plugins/SimulationView/SimulationView.py", line 347, in calculateMaxLayers
2018-01-17 23:11:10,227 - CRITICAL - cura.CrashHandler.__init__ [65]:     self._min_thickness = min(float(p.lineThicknesses[numpy.nonzero(p.lineThicknesses)].min()), self._min_thickness)
2018-01-17 23:11:10,227 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3.5/site-packages/numpy/core/_methods.py", line 29, in _amin
2018-01-17 23:11:10,227 - CRITICAL - cura.CrashHandler.__init__ [65]:     return umr_minimum(a, axis, None, out, keepdims)
2018-01-17 23:11:10,227 - CRITICAL - cura.CrashHandler.__init__ [65]: ValueError: zero-size array to reduction operation minimum which has no identity
2018-01-17 23:11:10,229 - DEBUG - UM.Controller.setActiveView [92]: Setting active view to SimulationView
2018-01-17 23:11:10,231 - ERROR - UM.Logger.logException [76]: Exception: An exception occurred while switching views: zero-size array to reduction operation minimum which has no identity
[... cut ...]
2018-01-17 23:11:10,236 - ERROR - UM.Logger.logException [80]:     self._min_thickness = min(float(p.lineThicknesses[numpy.nonzero(p.lineThicknesses)].min()), self._min_thickness)
2018-01-17 23:11:10,236 - ERROR - UM.Logger.logException [80]:   File "/opt/cura/lib/python3.5/site-packages/numpy/core/_methods.py", line 29, in _amin
2018-01-17 23:11:10,237 - ERROR - UM.Logger.logException [80]:     return umr_minimum(a, axis, None, out, keepdims)
2018-01-17 23:11:10,237 - ERROR - UM.Logger.logException [80]: ValueError: zero-size array to reduction operation minimum which has no identity
````

smartavionics, 2018-01-17:
I think this has been fixed today by 5c0e2d39a366d2daf78262f16826c27bf853a406 although it still takes forever to load and spews out lots of lines like this:
2018-01-17 23:24:15,505 - INFO - SimulationView.SimulationView.calculateMaxLayers [352]: Min thickness can't be calculated because all the values are zero

nallath, 2018-01-18:
This is why you shouldn't use nightly builds ;)

closed 2018-01-18 by nallath

BagelOrb, 2018-01-18:
I was using 3.1.0, not a nightly build.

BagelOrb, 2018-01-22:
So did you solve this? Or did you assume it was an issue of a nightly build?

Ghostkeeper, 2018-02-08:
I tested loading the attached g-code (in the 7z) and can confirm that it doesn't crash any more. It loads a bunch of vertical walls with rounded corners.

BagelOrb, 2018-02-08:
Indeed that was what the print looks like.

Good work!
:+1: