Ultimaker/Cura #3148: Warping prevention algorithm proposal

madanh, 2018-01-17:
This is a feature request.

# Background:
I have long parts delaminate from the bed on my (admittedly lousy) weistek WT150 printer, that does not have a heated bed. I just had an idea of a printing algorithm that should "theoretically" reduce this dramatically. Unfortunately I have absolutely no time to implement it as I'm writing a thesis, but I thought I need to share the idea as I believe some people might have similar issues.

# Theory:
## Problem 1: 
Warping is caused by thermal contraction of the solidified material as it cools down. As it is insulated and limited  from below the easiest way to relief tension is to curve up. This tension accumulates along long dimensions, i.e. along the filament.  When enough tension builds up ends of the filament (in the corners of the part), experiencing the most uncompensated tension, detach.

## Solution 1: 
If we avoid long stretches of filament the tension will be relieved before it can cause damage.
Print shell layers (especially the bottom ones) with regular detours inwards from the model surface. Adjust the infill accordingly. Admittedly it would cause dots on the surface, but some might prefer this to warping.

This is top view of the first two shells of the bottom layer, the detours could be shorter:
[image]

## Problem 2: 
With conventional printing top shell layers warp in the same general fashion as the bottom ones.Since they are stuck together this leads to accumulation of upwards force acting on the bottom layer at the same locations. 

## Solution 2: 
Now that we have the detours from solution 1 we can stagger their locations across the layers, then points that pull the strongest on the bottom layer will match with the points with zero pulling force from the next layer above it and vice versa.

Top: solution, bottom: illustration of the usual situation:
[image]

# Final notes:
If you could share this to other communities (e.g. RepRap), please do so.
If you choose to implement it please let me know (e.g. by commenting on this issue). 
In both cases a mention of my name (Hennadii Madan) would be nice. TIA

nallath, 2018-01-18:
One of the ideas i had for this is to use a checker board pattern for the first layer.

DDDirk, 2018-01-18:
I am not sure what you mean by problem 2, but the solution would prevent the formation of vertical fault lines. However, there is still the problem of vertical shrinkage.
But it could still help.
As for the dots on the surface, those might be prevented by not applying this to the outer shell. This would reduce the effect, but again, it might still help.
Also, you speak of the infill. Not using straight lines for that would probably also help.

madanh, 2018-01-18:
@DDDirk About infill - yes, but that would require quite a lot of effort to implement. Also just to make sure that we are on the same page: I mean not the infill of the model, but only the layer infill, which applies to bottom/top layers.

LOVE55, 2018-01-19:
This is a good solution to the glue. The glue is used on the Chinese side.

Ghostkeeper, 2018-02-06:
I remember a similar proposal that asked for these detours on the first layer, but going outside instead of inside for better adhesion vs. printing time. He called them "fingers". It was never implemented (yet).

One major disadvantage with your solution is that it will greatly reduce the surface quality. You're essentially introducing a lot of seams everywhere. We know that these seams also occur if they are on an inner wall (albeit less visible).
Another problem is with the implementation: You're not guaranteed that the polygon is big enough to hold a complete detour, so you must always be able to cut it short somewhere, but where and how to cut the detour short is difficult to compute.

For the rest it seems plausible to me, but a practical test can only show how effective it is.

BagelOrb, 2019-10-29:
The fingers were for adhesion, these detours are for reducing the accumulation of shrinking forces.