Ultimaker/Cura #3151: 关于回抽问题

LOVE55, 2018-01-18:
**Application Version**

**Platform**

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

不会英文，上面什么意思我不知道，以下面为准：
S3D以及CURA发现都有一个不足之处，就是回抽，回抽都是XYZ不移动的情况下执行的，造成FDM打印机不可避免的终点起疙瘩，要是能够有个参数能设置成回抽跟下一个空行程的代码同步进行就完美了，虽然会出现拉丝，这个可以作为一个选项，按需求选择，就这些。忘采纳

ChrisTerBeke, 2018-01-19:
Use English please so everyone can understand.

LOVE55, 2018-01-19:
@ChrisTerBeke I'm talking about the problem of withdrawing, CURA found the problem I have described this function today, but I didn't find it, thank you brother -- it is Machine Translation over, do not understand English, please forgive me

closed 2018-01-19 by LOVE55

LipuFei, 2018-01-19:
@LOVE55 你好，你说的“回轴”指的是反向推送材料吗？

Ghostkeeper, 2018-02-06:
I think you meant retraction? It's under the Material category.