Ultimaker/Cura #3152: First model layer raised when support uses towers and slicing mode is exclusive

smartavionics, 2018-01-19:
**Application Version**

master branch

**Platform**

Linux

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**

Load model that requires support, enable support towers and set slicing mode to exclusive.

**Actual Results**

First layer of model gets raised leaving just the towers and the skirt/brim on the build plate.

**Expected results**

First layer of model should be on the build plate (z = 0) irrespective of whether towers are being used or not.

**Additional Information**

This shows how the first layer only contains the towers/skirt
[image]

This shows what I would expect with the first layer of the model at z = 0 and the skirt going all the way round the model (towers disabled in this image)

[image]

jackha, 2018-01-22:
Does it happen on other models as well?

smartavionics, 2018-01-22:
I don't have any other examples right now but if my memory serves me right, I think I have seen other people reporting similar before.

Please see https://github.com/Ultimaker/CuraEngine/pull/685 which fixes this and another bug.

smartavionics, 2018-01-22:
You should be able to reproduce this using any model that has a region that needs support from the build plate and towers are enabled and the slicing tolerance is exclusive. Slicing tolerance exclusive will cause the first layer of the model to be removed but without the above bug fix, the model will not be moved down to the build plate because the tower will introduce support on the first layer so the layer will no longer be empty.

Ghostkeeper, 2018-02-08:
I tried creating a model to reproduce this, but didn't succeed in reproducing this bug. For me it looks like this: 
[image]

smartavionics, 2018-02-08:
I think it depends on whether the first layers or empty or not and that is influenced by the slicing tolerance. Some models (like my example) have an empty first layer when the slicing mode is exclusive. You need to find a model that behaves like that.

BagelOrb, 2018-02-28:
Could you please attach your problematic model?

If that's an issue, please attach the problematic part of that model (the bottom).

Are you sure the setting Enable Support Towers is relevant to reproducing the bug?

smartavionics, 2018-02-28:
Model can be found at https://www.dropbox.com/s/7ibxgk2nn2ks2i0/smartass-enclosure-flange.stl?dl=0

Yes, enabling the towers is required because that generates little blobs of support on the first layer at each end of the part.

BagelOrb, 2018-02-28:
Still can't reproduce. Maybe you should attach your project file. Then we are slicing with the same settings as well.

I'm trying to reproduce this on master, at the moment.

BagelOrb, 2018-02-28:
I can reproduce it if I lift the model up by 0.2mm.

I vote to approve the (modifed) PR.

smartavionics, 2018-02-28:
I've just noticed something that I haven't spotted before. If I click on that model, the move tool reports it to be at 0, 0, 7.85 when it is shown to be on the build plate. Why is z not zero?

BagelOrb, 2018-02-28:
Idk. Maybe you exported it with the object below the origin.

aschor, 2018-03-16:
hello, facing the same issue with cura 3.1.0 and cura 3.2.1. for ANY model used. I thought at first that it was the brims that were not calculated, but lowering the model changes it all (or choosing "middle" or "include", but not ideal)

closed 2018-03-27 by jackha