Ultimaker/Cura #3155: gMax 1.5+ Cura Profiles

gordo3di, 2018-01-19:
To get this out of the way, I will admit I am extremely new to github so no hiding that. That being said we manufacture and produce the gMax 1.5+ 3d printer and we have spent the last week finishing our Cura 3.1 profiles. We are very pleased with all the developments in Cura and would love to incorporate our configs to the repository.

After installing Git and Git desktop, cloning the repository, branching and trying to make a pull request I keep finding I don't have permissions. I'm sure this is common but after several hours of troubleshooting I'm stumped.

> Authentication failed. You may not have permission to access the repository or the repository may have been archived. Open options and verify that you're signed in with an account that has permission to access this repository.

I saw it says to "contact us" to add profiles and I'm not sure if that's Ultimaker or the github repository.

We plan on releasing the files on our fourm today so customers can install our profiles (we have a writeup explaining the install) and we would love to figure out how to use github properly for future updates.

Sorry again if i'm missing something obvious and thanks for all the hard work on Cura.

**Application Version**
Git desktop 1.0.12 
[gmax_cura_3.1_configurations_180119.zip](https://github.com/Ultimaker/Cura/files/1647019/gmax_cura_3.1_configurations_180119.zip)

fieldOfView, 2018-01-19:
The idea is that you "fork" the repository that you want to make changes to to your own account on github. Then you clone your own fork to your local computer/ You make your changes (preferably to a new branch), and push those changes to your fork (not the ultimaker original). Then from your fork, you can make a pull request.

I hope that gives you some search terms.

gordo3di, 2018-01-19:
Ah I guess I completely missed the concept of forking vs branching. Thanks for the quick reply. Working on it now.

closed 2018-01-21 by ChrisTerBeke