Ultimaker/Cura #3157: Support Z Distance not calculated

ChrisRiddell, 2018-01-20:
**Application Version**
3.1

**Platform**
Windows 10

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**
Open Cura and unhide the Support Z Distance or check the fdmprinter.def.json file

**Actual Results**
No matter what layer height you use the default Support Z Distance is 0.1mm even though the description says "..This value is rounded up to a multiple of the layer height."

**Expected results**
The value to be calculated to match it's description.

**Additional Information**

Ghostkeeper, 2018-02-07:
This doesn't reproduce for me in Cura 3.2. Perhaps we fixed it in the meanwhile?

My set-up is this: Ultimaker 2+, 0.4mm, Fine (0.1mm) quality. With the default settings the Support Z Distance is 0.1mm and after slicing it shows 1 layer of space between support and model. After setting the Support Z Distance to 0.5mm it shows 5 layers of space.

ChrisRiddell, 2018-02-07:
On a 0.15mm or the 0.06mm profile the Z distance in the input field stays as 0.1mm, Should it not update to the respected layer height value?

ianpaschal, 2018-03-26:
Doesn't reproduce in 3.2.1. Seems fixed.

closed 2018-03-26 by ianpaschal

hanturk, 2019-01-04:
It still not calculated in Cura 3.6 with standart fdmprinter.def.json. 
"support_z_distance":
                {
                   "label": "Support Z Distance",
                    "description": "Distance from the top/bottom of the support structure to the print. This gap provides clearance to remove the supports after the model is printed. This value is rounded up to a multiple of the layer height.",
                    "unit": "mm",
                    "type": "float",
                    "minimum_value": "0",
                    "maximum_value_warning": "machine_nozzle_size",
                    "default_value": 0.1,
                    "limit_to_extruder": "support_interface_extruder_nr if support_interface_enable else support_infill_extruder_nr",
                    "enabled": "support_enable or support_tree_enable",

In my winbo_superhelper105.def.json it is calculated like this:
"support_z_distance": { "value": "max((0.2 if(0.2%layer_height==0) else layer_height*int((0.2+layer_height)/layer_height)),layer_height)" }
Printers inherited from fdmprinter don't calculate. Nor top and bottom support values