Ultimaker/Cura #3158: CuraEngine Crash

ChrisRiddell, 2018-01-20:
**Application Version**
3.1

**Platform**
Windows 10

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**
Change any setting and windows pops up the CuraEngine crashed

**Actual Results**
CuraEngine crashed

**Expected results**
No crash

**Additional Information**
Model https://www.thingiverse.com/thing:1179160
Log [cura.log](https://github.com/Ultimaker/Cura/files/1648685/cura.log)

smartavionics, 2018-01-20:
Hi, I can slice that OK (and have previously printed that model, works great!). I think you will need to provide a profile so that we can see what your settings are.

ChrisRiddell, 2018-01-20:
[UM2+_V29D_Fixed.curaproject.zip](https://github.com/Ultimaker/Cura/files/1648710/UM2._V29D_Fixed.curaproject.zip)

It's a pretty modded profile with 2 lines of added gcode to the start gcode for a Z-offset seeing Cura does not have that setting.

smartavionics, 2018-01-20:
Sorry, I can't load that project as it makes my Cura blow up!

Devs, here's the log...

[cura.log](https://github.com/Ultimaker/Cura/files/1648718/cura.log)

ChrisRiddell, 2018-01-23:
> I think my cura could be blowing up due to your project using a material I don't have a definition for (colorfabb_petg_0.4_fast). Could you please create a new project that uses a stock material instead and then I may be able to load the project. Thanks.

I can do that when I get home but that profile is in my PETG repo on my github page.

smartavionics, 2018-01-23:
I have installed your material files as described in your README but they don't appear in the Cura materials dialog. Not sure how to proceed...

ChrisRiddell, 2018-01-23:
Hmmm it will only show on a UM2+ machine and you have to select colorfabb PETG from the materials dropdown.

But now i'm home I tested a normal profile and did not get this crash so I'm a little confused so maybe something in my profiles is making cura engine have a hissy fit.

smartavionics, 2018-01-23:
I have a UM2+ machine selected but the colorfabb materials do not appear in the material dropdown.

ChrisRiddell, 2018-01-23:
Hmmm that is interesting as the guide I made works on this computer I'm going to have to check on anther to see what's happening.

smartavionics, 2018-01-23:
Maybe it's because I am using the master branch?

ChrisRiddell, 2018-01-23:
Not sure I have no way of testing that as I have never been able to get cura to run from source on Linux or Windows even using the cura-build-environment repo

smartavionics, 2018-01-23:
Really? I have it installed on Linux using the cura-build-environment and it wasn't difficult to get it to work. I have it installed under /opt/cura and use a little script like this to run it:
````
#!/bin/sh

PATH=/opt/cura/bin:$PATH PYTHONPATH=/opt/cura/lib/python3/dist-packages /opt/cura/bin/cura $*

````

closed 2018-02-05 by ChrisRiddell