Ultimaker/Cura #3159: perspective vs orthographic view in CURA

darkdvd, 2018-01-20:
Please add to CURA a button to switch between perspective and orthographic view, because "perspective" is unusable to fine move a model.

Ghostkeeper, 2018-01-20:
Duplicate of https://github.com/Ultimaker/Cura/issues/2545.

The problem with this is that we'd need a different way to zoom as well.

closed 2018-01-20 by Ghostkeeper

darkdvd, 2018-01-20:
Excuse me Ghostkeeper but I don't understand why you need a diffrent way to zoom ?

Ghostkeeper, 2018-01-20:
Currently we zoom by moving the camera closer. The thing about orthographic projections is that things don't become larger when they get closer. So zooming like we do now has no effect at all.

darkdvd, 2018-01-20:
For perspective : move the camera closer
For othographic : change the camera view angle

darkdvd, 2018-01-20:
the bad idea in "CameraTool.py"

#   Note that zooming is performed by moving closer to or further away from the origin ("dolly")
#   instead of changing the field of view of the camera ("zoom")

fieldOfView, 2018-01-20:
And it is not just zooming that would need a “doover”; panning would need to change too.

krz4tw, 2019-03-13:
Was about to add a feature request on this as well. Id like to have it to be able to postion supportblockers precisely.

The zooming and panning should be quite easy to implement if you only enable orthographic in Top down, left, right,.... mode when you choose them via the little cube icons in the top right corner. Those 3-side projections are by far the most useful way for orthographic projection. I don´t need to orbit around in orthographic view.

jmsaltzman, 2019-03-13:
For reference, coming from an engineering and CG camera background: when you move the camera in and out, call it "dolly." When you change the view itself, call it "zoom." Makes these discussions much easier.

darkdvd, 2019-03-13:
And let's keep in mind that perspective view is about representing the object as we see it, while orthographic view is about representing the object as it is.

MoonCactus, 2019-03-19:
Also, having an ortho view makes it much easier to layout multiple parts on the printbed: checking the empty margin around multiple parts from above is not easy. Only layer views can help here but it is not convenient.
BTW, I would certainly not call this feature request a duplicate of #2545 which really is about measuring the part (as the titles tell: measurement vs. orthographic). They might arguably be related, but closing this one by redirecting it to the other (deferred) feature request is wrong imho.

Ghostkeeper, 2019-03-20:
It is considered a duplicate, because in an ideal world an issue addresses a problem rather than proposing a solution to the problem. That ticket addresses a need: Some way to measure stuff.

MoonCactus, 2019-03-21:
But in a real world, they are just two feature requests ("measure stls" is not a trivial functional duplicate of "have an orthographic projection"). Only the comments may converge: one of the way #2545 _could_ be implemented _might_ help implement this one as well, but not necessarily. This is far from obvious tousers that might keep asking for this projection (now, #5426 and possibly #4806 are real duplicates of this #3159 imo, but #2545 looks like another tool). No biggie :)

darkdvd, 2019-07-16:
Thank You for the **Orthographic** view !!!!!!

darkdvd, 2019-07-16:
[image]

fpdotmonkey, 2021-04-04:
For posterity, this feature was added with https://github.com/Ultimaker/Cura/pull/5768