Ultimaker/Cura #3160: Poor font quality in several UI parts

wavexx, 2018-01-22:
**Application Version**
3.0.3

**Platform**
Debian Unstable

**Qt**
5.9.2

**PyQt**
5.9.2

**Display Driver**
Intel(R) HD Graphics 520 (Skylake GT2)
kms driver on linux 4.14

**Steps to Reproduce**
I'm attaching a few screenshots of random parts of the UI to show how bad the font rendering is:

[image]
[image]
[image]

**Actual Results**
cura on debian depends directly on open-sans, which I guess is the font used throughout the UI.
However, open sans itself doesn't look as bad when used in other programs, including Qt5 programs. My guess here is that the bad appearance is due QML widgets not rendering/hinting/fitting the fonts properly?

I'm a developer myself, but I never used QML to actually comment.

**Expected results**
Granted, this is not a big issue, however it's appalling how bad the font rendering is.
Sometimes the descriptions of the parameters (the white-on-blue tooltip) is barely regible.

I'd love to try with a different font, but I see no way to change it in the prefs.

**Additional Information**

ChrisTerBeke, 2018-01-22:
Hi. We're aware of this issue. We're testing some other font options currently and hopefully will fix this soon.

wavexx, 2018-01-22:
On Mon, Jan 22 2018, ChrisTerBeke wrote:
> Hi. We're aware of this issue. We're testing some other font options currently and hopefully will fix this soon.

It would be nice to have the font to be configurable though.

I've chosen my main system font accurately and tuned the rendering
parameters via fontconfig, and I'd like to use it though-out.

ChrisTerBeke, 2018-01-22:
Not all fonts have the right sizes and weights available, so that would introduce uncertainty in creating our UI elements. I think it's better we settle on a font that renders well on all operating systems and common display sizes and drivers.

wavexx, 2018-01-22:
On Mon, Jan 22 2018, ChrisTerBeke wrote:
> Not all fonts have the right sizes and weights available, so that
> would introduce uncertainty in creating our UI elements. I think it's
> better we settle on a font that renders well on all operating systems
> and common display sizes and drivers.

With Qt this sounds like a strange statement to say, as most of the
widgets offer implicit autosizing and automatic layouts. This is also
the reason I'm surprised about the quality of the font rendering.

ChrisTerBeke, 2018-01-22:
Qt still relies on font files being available (or not). Not every font 'just works'. Just an example out of many: https://bugreports.qt.io/browse/QTBUG-45536.

Some font's we're researching don't have a semi-bold or italic variant. When a font doesn't have that, it's impossible for a GUI to make that up and just render it if you specify a weight that doesn't match any available font variant file.

We also don't use auto sizing and layouts that often as they're a bit too automatic and cause weird layouts.

wavexx, 2018-01-22:
On Mon, Jan 22 2018, ChrisTerBeke wrote:
> We also don't use auto sizing and layouts that often as they're a bit
> too automatic and cause weird layouts.

I'll wait and see then.

ianpaschal, 2018-01-23:
>With Qt this sounds like a strange statement to say, as most of the
widgets offer implicit autosizing and automatic layouts. This is also
the reason I'm surprised about the quality of the font rendering.

Qt does offer autosizing and automatic layouts but its font rendering can be incredibly unpredictable. Even using the same build on the same platform but different screens can radically alter how Qt renders a given font. In this case, using a system font by default is not a silver bullet.

As Chris said, we're testing other fonts to try and find a good solution that works every time on every platform on every screen. Likely is not possible 100% of the time, but can be possible _most_ of the time.

fieldOfView, 2018-01-23:
This has nothing to do with the specific font used.

Note that @wavexx is using 3.0. After the release of Cura 3, a big push was made to make sure that all text is rendered on whole pixel offsets. Qt has an issue with text that is placed on fractional positions, regardless of the font used. The issue is bigger for text that is rendered using Text.nativeRendering (QML Label does this by default).

@wavexx, please test Cura 3.1.

ChrisTerBeke, 2018-01-23:
@fieldOfView not sure if that's common knowledge. I was actually involved in making all those Math.floor() statements. Afaik the UI components shown above also have those....

fieldOfView, 2018-01-23:
It should be common knowledge in the Cura team, otherwise bad font rendering will be reintroduced.

> Afaik the UI components shown above also have those....

See the application version specified in the opening post. I must admit I too had to check when most of the Math.floor()s were introduced, but it was after Cura 3.0 (because they are not in the 3.0 branch, at least not for the two highlighted cases I checked from the opening post).

ChrisTerBeke, 2018-01-23:
Good point. @wavexx can you check in Cura 3.1?

wavexx, 2018-01-24:
Is there a PPA I can try without rebuilding from source? I tried with https://launchpad.net/~thopiekar/+archive/ubuntu/cura-master , but is simply segfaults on start.

fieldOfView, 2018-01-24:
No, there is an AppImage: http://software.ultimaker.com/Cura_open_beta/

wavexx, 2018-01-24:
On Wed, Jan 24 2018, Aldo Hoeben wrote:
> No, there is an AppImage:
> http://software.ultimaker.com/Cura_open_beta/

I tried Cura-3.2.0 beta. The rendering quality is acceptable now,
although still not at same level of quality of regular qt widgets.

So, consider this fixed.

But if I may add off record... I absolutely *hate* the interaction of
the various widgets of the UI. The list and table widgets feel sluggish
and do not respond at the same scrolling speed as the rest of the
system. I can see the pref dialog being _repainted_ while resizing the
window (and being repainted several times at that - as it too laggy to
repaint in real time). The setting list is ridiculously slow at both
repaint and scroll. Almost every interaction has noticeable lag.

The slicer and functionality is otherwise awesome so I say this from the
depth of my stomach, but the UI... I'm sorry... It feels on par with
electron-based apps, which is to say it's garbage.

fieldOfView, 2018-01-24:
If you want Cura to use the system default font family, you can do so with a theme. Download [this gist](https://gist.github.com/fieldOfView/3e94440bcc66acfd014dd2f71f6a4885) in place it in a subfolder of the themes folder in your configuration folder; Help -> Show configuration folder, go to "themes" and create a subfolder named eg "default-font".  

About your OT bit: we use Qt and lots of QtQuick Controls, through a python wrapper. It works well for me on Windows, and on Ubuntu. Doesn't always feel like a "native" application, but you have to draw the line somewhere when you want to publish a cross-platform application.

wavexx, 2018-01-24:
On Wed, Jan 24 2018, Aldo Hoeben wrote:
> If you want Cura to use the system default font family, you can do so
> with a theme. Download [this
> gist](https://gist.github.com/fieldOfView/3e94440bcc66acfd014dd2f71f6a4885)
> in place it in a [...]

That's a nice way around it, thanks for mentioning this.
I'll try later.

> About your OT bit: we use Qt and lots of QtQuick Controls, through a
> python wrapper. It works well for me on Windows, and on Ubuntu.
> Doesn't always feel like a "native" application, but you have to draw
> the line somewhere when you want to publish a cross-platform
> application.

I'd have to attribute the crappyness to QtQuick then, because I
regularly deploy QT programs (that I mostly do pyqt5) with regular
widgets, and at least under windows and linux they are excellent, even
when combined with the gtk2/3 platform theme.

Only on OSX I have some reserves in how they behave compared to Cocoa.

I'm aware of the limitations of using regular widgets in GL views, but
wow, what a difference.. :(

fieldOfView, 2018-01-24:
It probably does not help that Cura themes the widgets, instead of letting qt fall back to system widgets.

Ghostkeeper, 2018-02-13:
> So, consider this fixed.

We've fixed a few more of these in 3.3 too. I imagine it'll be an ongoing process as long as people keep forgetting to align everything to integer pixels.

closed 2018-02-13 by Ghostkeeper

ianpaschal, 2018-02-13:
> Doesn't always feel like a "native" application, but you have to draw the line somewhere when you want to publish a cross-platform application.

@fieldOfView: Not necessarily! I'm still toying with the idea of Node.js Cura as a personal project. Node has just-as-good-as if not better performance than Python, WebGL is very performant (especially for this sort of thing), and can re-use the CuraEngine C++ backend. Styling would then be done via CSS/LESS over QML (never thought I'd say this about CSS but thank god), also making it more accessible to users, I think. High DPI scaling becomes an issue of the past and everyone gets great text rendering. With the right font selector it's easy to use the system fonts as well.

@wavexx: When version 3.2.1 is shortly released there should be significant improvements. Apparently it doesn't fix all issues but it looks a _lot_ better than before.

wavexx, 2018-02-13:
On Tue, Feb 13 2018, Ian Paschal wrote:
>> Doesn't always feel like a "native" application, but you have to draw
>> the line somewhere when you want to publish a cross-platform
>> application.

I don't know, I mean, slic3r is mostly perl+wx, and the UI is vastly
superior for me, in just about every way. Zero lag, good quality font
rendering everywhere, no behavioral glitches. Even the GL view performs
better when rotating complex models. And I say that as somebody that
doesn't particularly like wxWindows. I'm sincerely blown away that
QML+C++ can get so bad.

Like I said, this feels more like an electron app rather than something
I'd expect from a desktop GUI.

> @wavexx: When version 3.2.1 is shortly released there should be
> significant improvements. Apparently it doesn't fix all issues but it
> looks a _lot_ better than before.

FIY, When I tested 3.2 two weeks ago it was already much better. I only
spotted the preferences dialog title, which was still badly rendered.

I'm currently using the last 3.1 for "production".