Ultimaker/Cura #3161: No Combing extra skin retraction over No Skin

ChrisRiddell, 2018-01-22:
**Application Version**
3.1

**Platform**
Windows 10

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**
Load 3DBenchy Turn off Combing Mode 

**Actual Results**
Their is a extra retraction points in the skin as it moves to the next line.

**Expected results**
For the skins I would of expected it to follow the same amount of reactions as No Skin.

I understand the extra retractions everywhere else that's the point of turning it off its great for runny plastics like PETG.

**Additional Information**
[image]

smartavionics, 2018-01-22:
Increase Retraction Minimum Travel ?

ChrisRiddell, 2018-01-23:
That seems to work to a point, to get it to the No Skin behavior it then starts effecting other areas where you want the combing off retractions.

Ghostkeeper, 2018-02-09:
The description says that if combing is off it should retract when moving to the next line, no?

It should only retract when moving through a wall. For some of these it is moving through a wall. For some it might be an edge case because the lines are very much tangent to the walls.

GregValiant, 2024-12-15:
Is this still an issue in current Cura versions (5.8 and up)?  Can this be closed?

closed 2024-12-30 by github-actions[bot]