Ultimaker/Cura #3162: Post Processing plugin bugs

ChrisRiddell, 2018-01-22:
This report covers a couple bugs in the post processing plugin

**Application Version**
3.1

**Platform**
Windows 10

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**
1. Add a post processing item 
2. Add a search and replace and save the gcode
3. Open the search and replace in the processing plugin _(suggestion more then a bug)_

> **example for 2:** 
> 
> Search 
> G92 Z0 ;reset zereo for Z Offset Hack
> 
> Replace
> G92 Z0 ;reset zereo for Z Offset Hack
> G1 X40 Z0.11 F4000 ;Z Offset Hack, also move the X to not smash the bed clip
> G92 Z0 ;reset zereo for Z Offset Hack

Actual Results
1. When auto slicing is disabled the prepare covers the tool icon
2. the exported gcode has the replace of the search duplicated 2 times 
3. the fields are of the input style _(suggestion more then a bug)_

> **example for 2:** 
> 
> G92 Z0 ;reset zereo for Z Offset Hack
> G1 X40 Z0.11 F4000 ;Z Offset Hack, also move the X to not smash the bed clip
> G92 Z0 ;reset zereo for Z Offset Hack
> G92 Z0 ;reset zereo for Z Offset Hack
> G1 X40 Z0.11 F4000 ;Z Offset Hack, also move the X to not smash the bed clip
> G92 Z0 ;reset zereo for Z Offset Hack

**Expected results**
1. not to cover tool icon
[image]
2. Only have it insert the replace once
3. the fields should really be textarea style _(suggestion more then a bug)_

**Additional Information**

Ghostkeeper, 2018-02-08:
1. This reproduces. It seems to only happen when auto-slicing is disabled.
2. I'm not able to reproduce this. I don't get the zereo reset thing, so I tried replacing `G92 E0` by `G92 E0\n;bla\nG92 E0`. It correctly duplicated every `G92 E0` with a `;bla` comment in between.
3. Indeed, I agree that it should be a multi-line text field, but we don't have an interface element for that available yet. At least, not in the styled setting list interface that's used in the post-processing window.

Ghostkeeper, 2018-02-08:
Fixed 1: https://github.com/Ultimaker/Cura/commit/e157d8e0455143cdb52afad430563352e0464a12

ChrisRiddell, 2018-02-08:
The zero thing was a hack for a z offset that I don’t need now because of the new plugin that’s available.

Maybe the doubling was because I was pasting a multi line code into the input and not using /n

closed 2018-02-25 by ChrisRiddell