Ultimaker/Cura #3163: Cura 3.1.0 does not accept PVA as material for Ultimaker 3

joheben, 2018-01-22:
**Application Version**
Cura 3.1.0

**Platform**
Windows 10 Home

**Qt**
Huh? I don't know that!

**PyQt**
Huh? I don't know that!

**Display Driver**
Not relevant

**Steps to Reproduce**
Opening two models (.3mf files), duplicating each (in total 4 objects).
Selecting Ultimaker3 as the printer.
Selecting PVA as material.

**Actual Results**
Cura 3.1.0 says that PVA is "incompatible material", which is WRONG (I print a lot on Ultimaker 3 with PVA, using previous Cura versions).

**Expected results**
That PVA is accepted as the material for Extruder 2.

**Additional Information**
I cannot locate the Cura log file, the instruction that it is at $USER/AppData/Roaming/cura/$CURA_VERSION/ is not valid ($USER is not set).

I cannot attach the .3mf Cura project file as suggested, and I do not have a ZIP program handy. I don't think it is important.

ChrisTerBeke, 2018-01-23:
Did you set the print core to type BB?

ChrisTerBeke, 2018-01-23:
Also, $USER is a variable, it's different depending on your username. Using $USER is a common way in development to indicate your user's home directory.

joheben, 2018-01-23:
Thanks for replying!

I did not realize that i had to change the printer core type from the 
default value (it is a new installation of Cura, new computer). That did 
solve the problem, in Cura 2.7 (I tried that earlier version, assuming 
it was a recent bug). Perhaps Cura should suggest to do that when a 
material is selected that does not work with the current core type?

(and I do know that $USER is a variable, tried it from the command line)

Kind regards,
// Johan

-- 
|
| CEO and Senior Project Manager
| InterNIT
|
| http://www.internit.se <Johan.E.Bengtsson@internit.se>
| E-contact: http://meetin.gs/meet/joheben @joheben 
http://google.com/+JohanEBengtsson
|
| Mobile: (+46)70 173 1000; Skype: joheben
| Address: Parkvägen 2, SE-972 53 Luleå, Sweden
---------------------------------------------------------------------------------

fieldOfView, 2018-01-23:
The proper path to use on Windows is:
```%APPDATA%\cura\``` (and the Cura version from there, but there is no environment variable for that, on any OS). ```%USERPROFILE%\AppData\Roaming\cura``` also works on most systems, but not on all (eg some systems which have the roaming profile moved to a network drive).

closed 2018-02-12 by Ghostkeeper