Ultimaker/Cura #3164: cura 3.1 don't open in macOS high Sierra 10.13.2

italogarcia2015, 2018-01-22:
**Application Version**
cura 3.1

**Platform**
macOS high Sierra apple 

**Qt**

**PyQt**

**Display Driver**

**Steps to Reproduce**
I download cura from oficial repository, Ultimaker, and after the download, I click int the icon and, it keeps jumping in the table but disappear and don't launch the program

**Actual Results**

**Expected results**

**Additional Information**

chrwang, 2018-01-23:
Cura 3.1 crashes on the first open after installing it; but works fine afterwards. It apparently doesn't get far enough in the start process to write any logs.

Reproducible after deleting the `~/Library/Application Support/cura` folder and deleting the Cura app from `Applications`, and recopying the app from the installer to the Applications folder.

Ghostkeeper, 2018-02-12:
We've seen the crash on first start-up on MacOS before (where it crashes the first time but runs fine the second time), but it doesn't even reach our code before it crashes so it is something with the way we build the application for MacOS. It also only happens on some computers and none of our development/testing computers. If someone has seen this before with cx_freeze we'd love to have your insight.

ChrisTerBeke, 2018-05-23:
No new info was added. Cura 3.3+ has better crash handling on boot. That should either fix the problem or give more insight. Please report back if you have more info from that. Closing for now.

closed 2018-05-23 by ChrisTerBeke