Ultimaker/Cura #3165: Initialization crash on multi monitor setup

Tomaschewski, 2018-01-23:
When I start up Cura 3.1, it crashes if the last window position was not on the main monitor (e.g. when "window_left" in the cura.cfg is set to a higher value than the main monitors width). As soon as I change the window_left value to zero, everything works fine again.

fieldOfView, 2018-01-23:
Do you have multiple GPUs, or a single GPU with multiple displays?

Tomaschewski, 2018-01-23:
It's a multi GPU setup, using the graphics chip on the CPU and a dedicated graphics card. :)

fieldOfView, 2018-01-23:
I can't help you there, because I don't have such a setup :-/

ianpaschal, 2018-03-05:
Since the author didn't use the issue template to provide relevant information, and I don't think any of us in Cura team can trouble shoot this (wrong set-up), I'm closing this.

ianpaschal, 2018-03-05:
Possible duplicate of #3170

closed 2018-03-05 by ianpaschal

PhotoJoeCA, 2024-02-24:
February 2024 Checking in using 5.4.0 and this is still an issue -- Multi-monitor system - Program hangs on launch.