Ultimaker/Cura #3166: OctoPrint plugin in 3.2

fieldOfView, 2018-01-23:
I just found out from the release notes for the 3.2 beta that OctoPrint is no longer distributed with Cura by default. A notice would have been courteous. 

Unfortunately, the version that is distributed through the plugin browser is the 3.1 branch of the plugin. Since the API was changed to support multiple buildplates (without changing the api_version I might add!), this 3.1 version does not work with Cura 3.2. Please update the plugin in the plugin repository with the appropriate version.

Might I suggest to put some order in the listed plugins? Alphabetical might be expected. Also please be so kind to spell my name correctly in the author field: fieldOfView with an undercast 'f'.

ChrisTerBeke, 2018-01-23:
You're totally right, you should have been informed! Apologies for that.

I'll raise a ticket for upgrading the plugin API version as well as updating the plugin on our server and give it priority.

Sorry again for the sudden changes.

ChrisTerBeke, 2018-01-23:
Devs: CURA-4857

ChrisTerBeke, 2018-01-23:
(bug in cura, 3rd_pary for the plugin upload)

ChrisTerBeke, 2018-01-25:
The plugin repo has been updated (v2 for Cura 3.1 and earlier does not contain Octoprint, v3 for Cura 3.2 has the OctoPrint plugin and several others that we have tested with this release so far).

@ianpaschal a change in the list order of plugins is suggested here, anything you can do in your refactor?

closed 2018-01-25 by ChrisTerBeke

fieldOfView, 2018-01-26:
Would it be an idea to make a github repo for the plugin repository? That way 3rd party developers can issue a PR to have their plugins (and metadata) updated. The cura-plugins repo could use branches for (Cura) API versions.

ianpaschal, 2018-01-26:
@ChrisTerBeke They are displayed alphabetically by default as part of the refactor. There's no custom sort/filter for now but I think that's OK.

@fieldOfView We've been talking a lot lately about what the plugin system might look like in the future and how it would integrate with a bigger "marketplace" with plugins, material profiles, etc. for download. This would likely include some sort of infrastructure similar to other package managers which would allow an author to publish a new of their package. A repo could be helpful in the mean time.

fieldOfView, 2018-01-27:
> The plugin repo has been updated 

Unfortunately that means that people who download the beta now don't have access to OctoPrint anymore, until a new 3.2 release is made (right?)

chking999, 2018-02-09:
Where do you get the Octoprint plugin from?

ChrisTerBeke, 2018-02-09:
Main menu -> plugins -> browse plugins -> OctoPrint connection -> install

Morgul, 2018-02-10:
Just came across this issue while trying to figure out how to get Cura working with OctoPrint again after an upgrade to 3.2, and I'd like to give a little bit of feedback to the team on this issue.

I read the release notes, and knew that OctoPrint was being moved to a plugin, but it was not clear that the plugin would not be _included_ in 3.2 by default. So my experience was updating, getting read to do a print, and then being confused when I couldn't connect to my printer. Until this point, I'd never used a plugin in Cura and was unfamiliar with the plugin browser, and mistakenly thought that `Settings -> Plugins` was it. It was only after finding the response above mine that I was able to get back up and running. Not a huge deal, but a little but of unintended friction, I'm sure.

Currently, the [release notes](https://ultimaker.com/en/products/cura-software/release-notes) say:

> **OctoPrint plugin added to plugin browser.** This plugin has moved to the plugin browser. This plugin enables printers managed with OctoPrint to print via Ultimaker Cura interface (version 3.2 or later).

They do not, however, say you will need to install the plugin if you were already using it. I would suggest changing the release notes to something like this:

> **OctoPrint plugin added to plugin browser.** This plugin has moved to the plugin browser. This plugin enables printers managed with OctoPrint to print via Ultimaker Cura interface (version 3.2 or later).
>
> _Note: If you used the OctoPrint plugin prior to 3.2, you will now need to install it. Simply go to `Plugins -> Browse plugins...` and click `Download`. Your previous settings will remain._

lenalebt, 2018-05-19:
That does not work or me. The "print with octoprint" button disappered, and I cannot get it to work again. Currently I need to save to file, upload to octopi manually and start the print job there, really annoying. Happy for input :).

Cura 3.3.1, Re-Installed the plugin multiple times, removed all settings as well (removing ~/.config/cura folder throwing away all settings). Octoprint reachable in the network (octopi.local), octopi 0.14.0, octoprint 1.3.8.

fieldOfView, 2018-05-19:
Is there a "Connect OctoPrint" button in the Printers pane of the preferences?
Can you post your logs (cura.log from ~/.config/cura) somewhere where we can have a look?

lenalebt, 2018-05-19:
I just found the issue. It forgot the API key for some reason. In case somebody else falls into the same trap, go to Preferences -> Configure cura -> Printers -> select your model -> Connect Octoprint and check that the API has not been removed. My bad, layer 8 problem. Thanks!