Ultimaker/Cura #3167: 3.2 Beta is out. will it be it safe to install it side by side with 3.1?

Thisismydigitalself, 2018-01-23:
Thank you all for Cura 3.2.
Will it be it safe to install it side by side with 3.1?

fieldOfView, 2018-01-23:
Yes.

Since Cura 2.6, new versions make a copy of the last version configuration they can find, and upgrade that copy (leaving the previous versions unaffected).

Thisismydigitalself, 2018-01-23:
Good to know but i though of asking again due to my (bad) experience back when i upgraded from 3.04 to 3.1Beta.

ChrisTerBeke, 2018-01-23:
Side by side is always OK. When it doesn't work for you, go back to 3.1 and you'll find nothing changed there. We always copy when upgrading to the newer version and only modify the copied data.

closed 2018-01-23 by ChrisTerBeke

Thisismydigitalself, 2018-01-23:
Once bitten, twice shy.
I will give it a try now after I've backup 3.1 and copied:
C:\Program Files\Ultimaker Cura 3.1\
C:\Users\~\AppData\Roaming\cura\3.1\

Thisismydigitalself, 2018-01-23:
Just installed 3.2B.
For those who want to keep 3.1, just click NO when asked to uninstall 3.1.
[image]

Installing 3.2B was fast and with no issue.
Loading time for 3.1R is: 32 seconds.
Loading time for 3.2B is: 9 seconds.

Impressive.
Will try it out and report back.
Thank you.

ChrisTerBeke, 2018-01-23:
Let's use this thread to report 3.1 vs 3.2 startup times so we have an average improvement overview for the final release.