Ultimaker/Cura #3168: Cura 3.2 Beta wont slice some 3.1 curaproject.3mf files.

Thisismydigitalself, 2018-01-23:
Yesterday I saved a 3.1 curaproject.3mf file which i can reopen with 3.1 and slice it, but it wont slice with 3.2 (not automatically nor manually). other 3.1 projects are sliced ok with 3.2, some not. unfortunately i cannot share those. i will try to find one which I can.

Thisismydigitalself, 2018-01-23:
Update: left 3.2 for few minutes while it was trying to slice and finally it did slice.
This is how it was stuck for few minutes - in this state:
[image]

I reckon it took about 3 minutes to slice this project where with 3.1 it takes few seconds.

Very strange.

Thisismydigitalself, 2018-01-23:
Sorry to say NO backward compatibility with 3.1 projects, at least i can't trust those.
just checked few projects, some which has NO top layers were opened in 3.2 but now WITH top layers, and such.

Appesteijn, 2018-01-23:
Is it possible for you to share the .3mf project files?

ChrisTerBeke, 2018-01-24:
We'll investigate that. That's why we do Beta's :)

Thisismydigitalself, 2018-01-24:
Thanks. just tested this again with a shared stl from thingiverse and i got the same result.
I can share the project but i guess you should experience the same with any stl. i sure like to know if so.

smartavionics, 2018-01-24:
As for the very slow slicing, can you make that project available or provide another one that exhibits the same problem? If not, could you please provide a screen shot? Anything?

Thisismydigitalself, 2018-01-24:
Hi, 
I don't know why but this does not happen anymore. it took about 3 minutes to open the file but only during first three attempts. from that moment on it opens as fast as with 3.1. I believe something was changed under the hood and I am not sure why but i think this might be related to the number of extruders. at least i saw something in the log file related to that.

So, about three times it took ~ 3 minutes to open, and from that moment on no problems opening it. 
I really like to share stuff with you guys and help you out but i feel uncomfortable sending files from some of my computers around here, others cases related to parts which are not mine to share.

smartavionics, 2018-01-24:
OK, well if it happens again and you can provide some info that would help a lot. Cheers.

Thisismydigitalself, 2018-01-24:
OK,
Just to show this is repeatable. just picked up an stl from thingiverse and saved it as 3.1 project.

3.1 project saved without top layers
[image]

3.2 project opened it and added top layers.
[image]

jackha, 2018-01-24:
@Thisismydigitalself I believe you that it is repeatable... but unfortunately not yet on my computer. I saved a project file with a cube without top layers in Cura 3.1, then load it in Cura 3.2 and it seems fine. I tried it again with clean user folders and it's also fine. 

Can you test this project file?

[UM3_cube20x20x20_c31.zip](https://github.com/Ultimaker/Cura/files/1659436/UM3_cube20x20x20_c31.zip)

jackha, 2018-01-24:
The slow slicing is also suspicious. It "should" currently only happen when a new feature, Tree Support, is enabled. It's in the Experimental section.

Thisismydigitalself, 2018-01-24:
Iopned UM3_cube20x20x20_c31 project on both 3.1 and 3.2.

On 3.1 - no top layers
On 3.2 - with top layers.

hmm..

jackha, 2018-01-24:
Ok really strange! Would you mind to empty the user folders of 3.2 and see what happens then? On Windows it's:
- $USER/AppData/Local/cura/3.2
- $USER/AppData/Roaming/cura/3.2

Thisismydigitalself, 2018-01-24:
After deleting these 3.2 folders the project now opens up and looks much different than before.
Now i get a topless cube with 7mm brim.

Thisismydigitalself, 2018-01-24:
Here's an important question?
If i change Profile and then revert back to the project profile which in this case is: FAST-02mm, - should i get the same settings?

If so, just know this is not the case.
I changed from FAST-O.2mm to FINE-0.1mm for example, and back to FAST-02mm,  and now 3,2 renders top layers..

What is going on here?

jackha, 2018-01-24:
By deleting the cura preferences the visibility of settings is also forgotten, but it's good to hear that it's something in the settings that caused this. Would you mind to share the 3.1 settings with us so we can have a look?

The project file I've sent does not contain any customized profiles, it's just a setting changed in the default (in this case Fast) profile. If you change profiles it should ask you if you want to keep or to discard those changed settings. If you say "Keep" on both switches, then you should get the same settings as before.

Thisismydigitalself, 2018-01-24:
You are correct about Keep and Discard. I admit i never really got the hang of these questions and got confused by those many times in the past - Including now.  I want to say that for me and as i always thought so is that projects should be like Zip file. whenever you open it you get the exact same folder's structure and files. With CuraProjects haowever I was never a happy camper. I always got weird results. I believe out of Cura issues which are far less than the benefit of using it, the Project's behavior is what i could never really trust. Just my own personal experience.

jackha, 2018-01-24:
Maybe the biggest benefit of project files is that it is able to help us debug (remote) users problems really well ^^

Thisismydigitalself, 2018-01-24:
Projects contains models and who know what else. Models are not always mine to share and I don't like sharing personal information - Something which some would consider crazy but that's (just?..) me :)

jackha, 2018-01-25:
Actually project files are "just" zipped folders with a defined content that you can check yourself: the 3d model in a specific format and current Cura settings (which profiles, what printer, what settings were manually changed etc). It is true that the models are not always yours to share, so be careful about that.

Thisismydigitalself, 2018-01-25:
So why do i always have issues with projects across Cura versions?
FWIW,  i never trusted projects in a way like reopening a week old project and send to printer - I always, but always have to double and triple check all settings before printing.  To this moment i don't understand why 3.2 did not open my 3.1 projects (some do not open at all) looking the same as they were saved with 3.1.  why deleting 3.2 folders and starting 3.2 from "scratch" resulted in a completely different outcome - one which is closer to what i would expect.

jackha, 2018-01-25:
(For now) it's wise to check everything before printing from project files that are across Cura versions. The problem is that we change the structure of the internals of cura settings and stuff and then the "old" config files should still load correctly. Known issues regarding this are being fixed.

Thisismydigitalself, 2018-01-25:
For the first time ever i spent some time tying to understand my profile's issues.
What striked me was the fact that i noticed that with other printers, unlike my Custom FDM printers, there is no EXTRUDER pull down menu. my printers are configured as Custom FDM which by default sets EXTRUDER1 so in profiles you get 2 side by side but different settings (!?).

When i change my settings and then update my profile (clicking on the star symbol) i see that the Global settings is updated but not the EXTRUDER1 settings. I also noticed that other printers from the list of printers which I know has only single extruder do NOT have the option to chose how many extruders there are.

Please explain.

jackha, 2018-01-29:
Don't all printers have the "Number of Extruders" setting in Machine Settings? I tried Creality CR-10, Mendel90, Cartesio and they all have the setting. 

The settings changed regarding to profiles depend on the setting itself. Some are global (e.g. Layer Height, Build Plate Temperature), while others can be set per extruder (e.g. Infill Density). When a profile is saved, it saves the global changes as well as the per extruder changes. When changing a global setting, then the extruder changes remain empty. Does this clear things up?

fieldOfView, 2018-01-29:
> Don't all printers have the "Number of Extruders" setting in Machine Settings?

No, only machine that define extruders.

Thisismydigitalself, 2018-01-30:
> The settings changed regarding to profiles depend on the setting itself. Some are global (e.g. Layer Height, Build Plate Temperature), while others can be set per extruder (e.g. Infill Density). [...]

Makes sense. thanks. i need to spend time checking if this is the case on my side but i just don't have spare time to deal with all of this. What i'm trying to do is find out why my Cura wont save/open projects. this drives me crazy cause each time i need to print copies i need to go over all settings again and this is a BIG NO NO.

jackha, 2018-01-31:
Is it possible to share the project files? And then pictures like above of what you expect and what you actually see. I can have a look what happens on my machine.

Thisismydigitalself, 2018-01-31:
https://github.com/Ultimaker/Cura/issues/3201

Ghostkeeper, 2018-02-26:
Not able to reproduce this yet. I've loaded Jackha's project in 3.2 and it had no top layers (as expected, the Top Layers setting was set to 0). I've used 3.1 to create my own project file and it also loaded as expected with infill set to 42 and layer height set to 0.1337.

ianpaschal, 2018-08-01:
This is like 6 months old with no new activity and we're unable to reproduce it. Sorry folks, but this case has gone cold...

closed 2018-08-01 by ianpaschal