Ultimaker/Cura #3169: Zooming stuck bug

ovidake010, 2018-01-23:
**Application Version**
3.1.0

**Platform**
Windows 10

**Display Driver**
Nvidia 388.31

**Steps to Reproduce**
Pressing **_space+left click_** to move camera and zoom,

**Actual Results**
The zooming function remain stucked, like the combo buttons are pressed, but they aren't. 

**Expected results**
Once i released the combo buttons, cura needs to stop zooming. It is working for a few times, but whenever Cura want, it remain stucked.

fieldOfView, 2018-01-23:
space + left click? Do you mean shift + left click?

ovidake010, 2018-01-24:
shift and left click is to move the camera on one axis (like 2d), space+ left click is to move the camera "in space" like 3d.

jackha, 2018-01-24:
I didn't know this feature existed XD. Can you provide us with the log files?

fieldOfView, 2018-01-24:
I just checked the code. The idea is you press the space bar and then move the mouse without clicking. This zooms the camera, but does not rotate the view. The Space + drag behavior is a bit hard to control (and IMHO a bug by itself)

jackha, 2018-01-24:
I heard the space + drag is intended for computers without a right mouse button.

ovidake010, 2018-01-24:
Thank you for your answers guys.
So it is a way to solve this bug? i mean, it's annoying to arrange 3-4 objects to print, and then suddenly the zoom is stuck and need to restart cura :( ...

jackha, 2018-01-25:
I can't reproduce this, but what do the logs say? ^^

Tom5ive, 2018-02-03:
I can confirm that this bug is effecting me on OS X High Sierra also .

Ghostkeeper, 2018-02-26:
Jackha tested on Windows 10. Not happening on Ubuntu 17.10 either. The button doesn't get stuck however often I try... Moving the mouse out of the window while dragging, moving it onto a text box, etc. all still seems to work properly.

ianpaschal, 2018-02-26:
I've tested v3.2.1:

- on macOS High Sierra
  - with MBPr keyboard+trackpad
    - with empty scene
    - with populated scene
  - with external keyboard+mouse
    - with empty scene
    - with populated scene

...and can't reproduce it.

My guess is that it was fixed in 3.2 or 3.2.1, so we should close this and consider it fixed. If @ovidake010 and @Tom5ive are still experiencing it after upgrade, we can re-open, but I bet not.

closed 2018-02-26 by ianpaschal

Ghostkeeper, 2021-11-05:
At this point, having closed this issue 3 years ago, I think it's better to make a new report.

I tried to reproduce this issue but couldn't get it to stick. However I didn't keep trying for a couple of hours either, like you did.

Hunkoys, 2022-12-17:
2022. Still an issue

ShaneCroke, 2024-06-22:
2024 still an issue