Ultimaker/Cura #3170: Application crash on start up - Ultimaker Cura 3.2 BETA

BenMarolt, 2018-01-24:
**Application Version**
Ultimaker Cura 3.2 BETA
Downloaded from Ultimaker's website

**Platform**
Windows 7 Ultimate 64-bit

**Display Drivers** (2 video cards in this system)
NVIDIA GeForce GT 630 23.21.13.8813
NVIDIA GeForce GTX 960 23.21.13.8813

**Steps to Reproduce**
Installed Ultimaker Cura-3.2.0-BETA-win64.exe with standard options to install arduino drivers and visual studio 2015 redistributable, but unticked the check box for opening STLs with Cura, and installed to secondary hard drive. (I did not uninstall Cura 3.1.) (Also tried installing to the default program folder, and got the same behavior failing to start.)

**Actual Results**
While starting up the program, the splash screen progresses through the start up steps until Setting up scene... Then an error pops up:
cx_Freeze: Python error in main script (traceback unavaulable)
Error in sys.exceptionhook.
Exception: list index out of range
Original Exception: max() arg is an empty sequence
[image]

**Expected results**
The application window should appear after the splash screen.

**Additional Information**
I previously had cura 2.4, 2.6, and 2.7 installed, but uninstalled them. I was using a custom printer definition with 3.1. I did try modifying the related files in user and the json that I had added to definitions to change all the versions from 2 to 3, which did allow the 3.1 BETA to start, but there were no printers under manage printers. Although the add printer feature would allow choosing a type of printer to add, after confirming, the window would close out without adding anything to the list of printers under manage printer. So without being able to add a printer, I couldn't do much.
   In an attempt to avoid incompatible old definitions. I uninstalled 3.2 BETA.  I renamed the 3.1 program directory and renamed the AppData\Local\cura folder. Then tried installing to the default directory on "C:\Program Files", but I really wanted it to be on "V:\Other Program Files" on a secondary hard drive where there is more storage space available.  In between attempts to uninstall and reinstall the 3.2 beta, I deleted the cura folder from appdata which would be AppData\Local\cura\3.2\cache\definitions\3.2.0-BETA\ with only two files:
custom
fdmprinter
[3.2.zip](https://github.com/Ultimaker/Cura/files/1658245/3.2.zip)

Any tips on how I can get this running properly? I'm ready to re-create my custom printer definition, but even with it removed, it seems to me like the custom and or fdmprinter definitions as installed are potentially causing a problem. Anything in the windows registry I should clear out?  I've been anxious to try the adaptive layer heights since seeing the Variable layer heights #647 pull request on github in November.

jackha, 2018-01-24:
@BenMarolt First of all, thank you a lot for the very detailed information!

First let's see if the old data is causing this problem. You can empty these folders, but leave the folders there. This will prevent Cura from trying to upgrade files from an older version (which could be the problem here).

- AppData\Local\cura\3.2
- AppData\Roaming\cura\3.2

In Cura we don't use any windows registry. I think there are only file associations here for Cura.

Let us know if it helped.

ctzsnooze, 2018-01-24:
Same issue on a mac; 3.1 opens fine, 3.2 beta vanishes on startup at the end of Loading Machines... no dialog, just vanishes.

Console says com.apple.xpc.launchd[1]: (nl.ultimaker.cura.1617632[5343]) Service exited with abnormal code: 1

Looks like an issue parsing a machine.

jackha, 2018-01-24:
@ctzsnooze Sorry to hear you have the problem. Could you provide us with the 3.1 cura settings dir so we can reproduce it? Also if do don't mind, you can see if you get to to work if you delete the cura preferences dir. (Clear Cura preferences see: https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations)

BenMarolt, 2018-01-24:
I see I failed to remove the old data in Roaming.
[Roaming3.2.zip](https://github.com/Ultimaker/Cura/files/1659930/Roaming3.2.zip)
[Roaming3.1.zip](https://github.com/Ultimaker/Cura/files/1659931/Roaming3.1.zip)
[Roaming-cura.zip](https://github.com/Ultimaker/Cura/files/1659936/Roaming-cura.zip)

BenMarolt, 2018-01-24:
OK, well, I've moved everything I can think of over from the 3.1 settings into 3.2 Beta and its running fine. I'm not sure which profile or setting was causing the problem, but this was probably due to old data from me messing with profiles. Even the machine definition json I hacked together is working in 3.2 Beta. It is great to have adaptive layer height (without resorting to slic3r)

printingotb, 2018-01-24:
I had the same issue and solved it the same way by clearing both local and roaming folders.
However, I cannot use my old profiles even though I exported them and try to import them in the 3.2beta.
Could you try to open an older version of cura, if you have any, and migrate a few profiles to the new version? I
I opened a new issue about this 40 min ago.
https://github.com/Ultimaker/Cura/issues/3177

BenMarolt, 2018-01-25:
I was able to migrate quality profiles and materials using the export from 3.1 and the import on 3.2 Beta without any problems.

As far as moving printers to 3.2 beta, it wasn't as simple.  I have a printer def.json file (based on "custom" but with just one extruder, different default dimensions and some disallowed areas) that inherits from "fdmprinter", which I just copy pasted from Program Files\Ultimaker Cura 3.1\resources\definitions to Program Files\Ultimaker Cura 3.2\resources\definitions and it showed up as an option to create a new printer under "Manage Printers". I had to copy paste my start code and end code into the new printer in 3.2, since those weren't part of my def.json file.  I was expecting problems seeing that the the extruder settings are separated to another tab in the UI, but it worked fine with the same def.json, even though I originally made for Cura 2.7, and it still carries the version 2.

I did not have much luck copying a printer machine instance (*.inst.cfg) from the %AppData%\cura\3.1\user directory to its 3.2 counterpart, but didn't try too hard since I only have one printer instance to deal with and adding a new one from the def.json worked easily enough for me.

ctzsnooze, 2018-01-25:
I could open Cura 3.2 by deleting everything in Users/me/Library/Application Support/cura/3.2.

Unfortunately that meant I lose my custom printer, materials, profiles, etc.

It's annoying that I can't export my current custom printer (not one Cura supports) from 3.1 and then import it into 3.2.  The same thing happened when updating from 3.0 to 3.1.

BenMarolt, 2018-01-25:
The materials and profiles import and export on Windows works reasonably well for me. Maybe more related to [printingotb's issue ](https://github.com/Ultimaker/Cura/issues/3177) and more of a feature request than a comment on this issue, but could there be a similar import/export for printers?

BenMarolt, 2018-01-25:
Does anyone think I should close this issue, yet? The workaround of removing the conflicting data from the cura\3.2 directory solved it for me for now, and that seems like an appropriate way to deal with it on a beta, but it would be a tad irritating if the problem of crashing due to old data persists in a stable release.

jackha, 2018-01-25:
Leave it open. I created a ticket so we can have a deeper look at it. It is annoying for "upgraders" if it doesn't work out of the box.

Devs: CURA-4868

LipuFei, 2018-01-29:
Hi @BenMarolt , I see you have an ANET A8 printer. Do you have some sort of definition files downloaded somewhere for ANET A8?

jackha, 2018-01-29:
@LipuFei did you reproduce it? Here we're having trouble reproducing the problem with the provided files.

Ellecross, 2018-01-29:
@LipuFei I saw that as well because in the zipped 3.1 Roaming folder, there are Definitions for Anet A8 Benz custom printer. Here's the file @BenMarolt originally uploaded: 
[Roaming3.1.zip](https://github.com/Ultimaker/Cura/files/1673120/Roaming3.1.zip)

I see a lot of errors in the logs when opening, but it does not block Cura from starting. But I get this error in the logs: 

2018-01-29 10:59:52,858 - ERROR - UM.Logger.logException [80]: Exception: When trying to deserialize Anet A8 Benz, we received an unknown ID (Anet A8 Benz_user) for container

BenMarolt, 2018-01-29:
@Ellecross I have an Anet A8 but I've modified it heavily. Different dimension heat bed and changed some of the electronics, so my definition file isn't likely to be super useful for anybody else. It's based on the custom printer definition with some of the default values changed. I did get this working after getting around the issues with other old data that were crashing Cura. I'm including a zip of it here for reference, but be warned I run a really hot heatbed for ABS, so this may be dangerous to use  (power supply and or wiring may actually catch fire) on a stock Anet A8 without modifying at least the start code:
( This would live in "Program Files\Ultimaker Cura 3.2\resources\definitions\" )
[AnetA8Benz.def.json.zip](https://github.com/Ultimaker/Cura/files/1674611/AnetA8Benz.def.json.zip)

Anybody reading this thinking they can use it with their own printer: **_make sure to modify this to match your printer before use!_**

BenMarolt, 2018-01-29:
@jackha Sorry, I should have made a zip of the entire cura AppData directories before I started troubleshooting on my own. There were initially folder for 2.4, 2.6 and 2.7 and I wonder if they were also involved, but I deleted them early on and I don't have a backup.

jackha, 2018-01-30:
@BenMarolt I also didn't think of it... colleagues are looking at it. Let us know if you can re-reproduce it ;-)

LipuFei, 2018-01-30:
@BenMarolt I have checked the files you have uploaded. Unfortunately, there are some incorrect configuration files and we would suggest to have a clean install. If this is OK for you, please follow the steps below:
1. Backup (optional) and remove the following two directories:
- C:\Users\<your-username>\AppData\Local\cura\3.2
- C:\Users\<your-username>\AppData\Roaming\cura\3.2
2. Create this directory so it is empty:
- C:\Users\<your-username>\AppData\Roaming\cura\3.2
2. Start Cura 3.2 BETA. You shall have a clean CURA 3.2-BETA installation.

BenMarolt, 2018-01-30:
@LipuFei Yup, 3.2 Beta has been running fine for the past 6 days since I did similar. Although I also renamed the 3.1 directories during install of 3.2 beta to prevent the installer from importing anything from there, which may have been an extra unneeded step. That's essentially what worked for me last Wednesday, except I didn't get backups of the AppData at first. ( ...And my Appdata is not actually on C: )
Uninstalling/reinstalling 3.2 beta since then has been error free for me.
 Going forward I'll try to remember to zip up both local and roaming AppData before upgrading/installing new versions/beta versions of Cura, in case anything goes wrong that might be useful to reproduce later.

BenMarolt, 2018-01-30:
@jackha Thanks, my attempts to reproduce the error have come up with nothing. 3.2 works for me now, only having issues with experimental settings, which is to be expected. Pretty sure my issue with installing and starting up Cura 3.2 beta was limited to the old data from before version 3.

jackha, 2018-01-31:
We've made crashes like this more graceful. If something goes wrong during startup, it's probably due to some configuration error (which may be the result of upgrading older data). It now offers you to reset the configuration and start clean.

We'll see if there are other people who have the same problem as you had and that can possibly fixed as well (so you don't have to start clean).

BenMarolt, 2018-01-31:
Cool, the option to choose fresh config like that sounds like a pretty good solution.

davidwyl, 2018-02-04:
current;y im at 3.0.4, i unable to upgrade to 3.1 (it will crash before it start), today i try 3.2 beta im getting the same python error...
i deleted all the 3.2 data in Local & roaming, 3.2 run without problem but it start fresh without my printers & profiles.
Is there a way to install 3.2 beta without loosing all my profiles and printers?

jackha, 2018-02-05:
@davidwyl we try to fix common "upgrade errors". If you're willing to share your 3.0.4 preferences folder along with the logs, we can have a look at it.
Currently we've implemented a "catch all" - we actually don't know what happened that caused this.

davidwyl, 2018-02-05:
@jackha here is the preferences and log https://drive.google.com/file/d/19jK2s47Uzz5Ke4Jq5uYuSI6heE0cTAkT/view?usp=drivesdk

jackha, 2018-02-05:
@davidwyl thanks. Can you include the "3.0" folder as well? It contains the configured printers and settings that it's going to upgrade.

davidwyl, 2018-02-05:
@jackha https://drive.google.com/file/d/1yXPpAON8Em21pK8wsP_e1fDZyrYCE0cs/view?usp=drivesdk

jackha, 2018-02-06:
Thanks :-) We'll have a look.

RichardSerton, 2018-02-07:
I'm seeing this problem in the non-beta 3.2 version for Mac OS as well.   Cura starts and when I select a printer (Monoprice Mini), the application just vanishes.

The cura.log shows:
2018-02-07 16:54:18,410 - CRITICAL - [(140736075109184)-MainThread] cura.CrashHandler.__init__ [64]: An uncaught error has occurred!
2018-02-07 16:54:18,411 - CRITICAL - [(140736075109184)-MainThread] cura.CrashHandler.__init__ [67]: Traceback (most recent call last):
2018-02-07 16:54:18,413 - CRITICAL - [(140736075109184)-MainThread] cura.CrashHandler.__init__ [67]:   File "/Users/ultimaker/build/3.2/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 800, in event
2018-02-07 16:54:18,415 - CRITICAL - [(140736075109184)-MainThread] cura.CrashHandler.__init__ [67]:   File "/Users/ultimaker/build/3.2/build/inst/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 339, in event
2018-02-07 16:54:18,417 - CRITICAL - [(140736075109184)-MainThread] cura.CrashHandler.__init__ [67]:   File "/Users/ultimaker/build/3.2/build/inst/lib/python3.5/site-packages/UM/Event.py", line 189, in call
2018-02-07 16:54:18,418 - CRITICAL - [(140736075109184)-MainThread] cura.CrashHandler.__init__ [67]:   File "/Users/ultimaker/build/3.2/build/inst/lib/python3.5/site-packages/cura/Settings/ExtruderStack.py", line 104, in <lambda>
2018-02-07 16:54:18,420 - CRITICAL - [(140736075109184)-MainThread] cura.CrashHandler.__init__ [67]:   File "/Users/ultimaker/build/3.2/build/inst/lib/python3.5/site-packages/cura/Settings/ExtruderManager.py", line 528, in updateMaterialForDiameter
2018-02-07 16:54:18,422 - CRITICAL - [(140736075109184)-MainThread] cura.CrashHandler.__init__ [67]: TypeError: type SettingFunction doesn't define __round__ method
2018-02-07 16:54:18,424 - CRITICAL - [(140736075109184)-MainThread] cura.CrashHandler.__init__ [64]: An uncaught error has occurred!
2018-02-07 16:54:18,425 - CRITICAL - [(140736075109184)-MainThread] cura.CrashHandler.__init__ [67]: Traceback (most recent call last):
2018-02-07 16:54:18,427 - CRITICAL - [(140736075109184)-MainThread] cura.CrashHandler.__init__ [67]:   File "/Users/ultimaker/build/3.2/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 800, in event
2018-02-07 16:54:18,429 - CRITICAL - [(140736075109184)-MainThread] cura.CrashHandler.__init__ [67]:   File "/Users/ultimaker/build/3.2/build/inst/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 339, in event
2018-02-07 16:54:18,431 - CRITICAL - [(140736075109184)-MainThread] cura.CrashHandler.__init__ [67]:   File "/Users/ultimaker/build/3.2/build/inst/lib/python3.5/site-packages/UM/Event.py", line 189, in call
2018-02-07 16:54:18,432 - CRITICAL - [(140736075109184)-MainThread] cura.CrashHandler.__init__ [67]:   File "/Users/ultimaker/build/3.2/build/inst/lib/python3.5/site-packages/cura/Settings/ExtruderStack.py", line 104, in <lambda>
2018-02-07 16:54:18,434 - CRITICAL - [(140736075109184)-MainThread] cura.CrashHandler.__init__ [67]:   File "/Users/ultimaker/build/3.2/build/inst/lib/python3.5/site-packages/cura/Settings/ExtruderManager.py", line 528, in updateMaterialForDiameter
2018-02-07 16:54:18,436 - CRITICAL - [(140736075109184)-MainThread] cura.CrashHandler.__init__ [67]: TypeError: type SettingFunction doesn't define __round__ method

Hope this helps.

fieldOfView, 2018-02-07:
See https://github.com/Ultimaker/Cura/issues/3246, which also includes a fix

ChrisTerBeke, 2018-07-30:
Fixed in  #3165

closed 2018-07-30 by ChrisTerBeke