Ultimaker/Cura #3171: Retracts on tower

karabas, 2018-01-24:
**Application Version**
CURA 3.1

**Platform**
win10

**Steps to Reproduce**
Dual printing on cyclop hotend, 0.8mm nozzle

**Actual Results**
when printing prime tower it does retract on EVERY print move. It slows down total speed and generates upward strings like "hedgehog"

**Expected results**
as in older versions of cura. No excessive retracts.

jackha, 2018-01-24:
You mean during the print of the prime tower or in a print that contains a prime tower? Could you provide a project file to show this? That would make reproduction a lot easier.

karabas, 2018-01-29:
I mean during the print of the prime tower.

karabas, 2018-02-01:
Sorry I use 0.8 nozzle and default retract distantace 1mm too small. 
Better to calculate default from nozzle diameter *  2 or something

closed 2018-02-01 by karabas

Ghostkeeper, 2018-02-26:
I found that on Ultimaker 2+ using 0.8mm nozzles and PP material, the faster quality profiles had a Retraction Minimum Travel setting to be lower than the line width, which causes retraction between every skin line. After discussing that with our Materials & Processing team we've decided to reset that setting to the default (line width * 2).

commit 91d271b9 (2018-02-26): Reset UM2+ PP profiles retraction minimum travel

These settings were set to be lower than the line width. I'm resetting them so that they will again be 2 times the line width. Otherwise we'll get retractions between every skin line which is really bad for PP.

Fixes #3171.