Ultimaker/Cura #3172: [PostProcessingPlugin - PauseAtHeight] Please add "pause before layer N" in addiction to height in mm (and custom GCODE too)

noisemaker00, 2018-01-24:
The `PauseAtHeight` plugin is cool, but it will be more useful if you can set the layer number BEFORE which the printer should pause and wait.
When you embed extra parts (like nuts) you can check in the "Layer view" the layer which will cover the part, so you must set a pause before that layer. This is faster and easier than calculate the height.

It will be cool if you can also set a custom GCODE script to replace the one generated from the plugin itself.

Ghostkeeper, 2018-02-12:
Something similar to TweakAtHeight would be nice to have, where you can specify a layer number.

Ghostkeeper, 2018-02-12:
I've implemented this for Cura 3.3, but I didn't port it for the Repetier or BQ versions yet. Kind of difficult since we have 3 implementations of the same thing...

commit 75265911 (2018-02-12): Add option to pause at layer

Fixes #3172.
closed 2018-02-12 by Ghostkeeper